ROBOTIS-GIT/open_manipulator #137: Thingiverse missing right RAIL_BLOCK stl

nxdefiant, 2019-12-17:
Hi,

it seems to me that the Open Manipulator [package on Thingiverse](https://www.thingiverse.com/thing:3069574) is missing the stl file for the right rail block? It has the "RAIL_BLOCK.stl" which I believe is the left one.

rjshim, 2019-12-17:
@nxdefiant  

You can just print out two of that file. It should be up down symmetric.

Ryan.

closed 2019-12-17 by rjshim

nxdefiant, 2019-12-17:
it doesn't look very symmetric to me..

Also flipping one axis is not an option because of the nose on the edge.

rjshim, 2019-12-17:
@nxdefiant 

Maybe the part you say is not the RAIL_BLOCK.stl file but something else. Check the [part list](http://emanual.robotis.com/docs/en/platform/openmanipulator_x/assembly/). 

Ryan

nxdefiant, 2019-12-17:
sorry my fault. Its the RAIL_BRACKET was referring to.

nxdefiant, 2019-12-17:
This:
[image]

rjshim, 2019-12-17:
@nxdefiant 

Thank you for the report. I just updated the uploadings. Download them and let us know if you still have other issues.  :)

Ryan

nxdefiant, 2019-12-17:
That looks good, thanks.