ROBOTIS-GIT/open_manipulator #138: IKFast Kinematics Solver

nxdefiant, 2020-01-04:
Hi,

could you please provide an IKFast solver for the default manipulator configuration? (The one in the urdf).

When dragging the goal state marker in rviz the KDL ik solver in many cases fails to calculate the inverse kinematic for the 4d open manipulator. The workaround is to set position_only_ik=true with the disadvantage that is prevents to set any angle. An IKFast solver will increase the usability a lot.

I successfully created an ikfast solver using OpenRave with iktype=TranslationZAxisAngle4D. Moveit currently needs a little [patch](https://github.com/ros-planning/moveit/pull/1823) but that should be solved in the near future.

rjshim, 2020-01-09:
@nxdefiant 

You can add your code to the [file](https://github.com/ROBOTIS-GIT/open_manipulator/blob/master/open_manipulator_libs/src/kinematics.cpp) and use it if needed. We do not have any plans to update our package yet but we will let you know if we do. 

Regards,
Ryan

closed 2020-01-09 by rjshim

nxdefiant, 2020-01-09:
Interesting, can you please point me to the documentation on how to use this kinematics solver with moveit?

rjshim, 2020-01-09:
Please leave your question at the [link](https://github.com/ros-planning/moveit).

nxdefiant, 2020-01-09:
I don't think that it is the job of moveit to document on how to add a custom open manipulator kinematics solver. 

I guess I better stay with the openrave/ikfast solver. It works fine, just a pita to generate, which is why I have attached it. Maybe someone else finds it useful.

[ik.cpp.zip](https://github.com/ROBOTIS-GIT/open_manipulator/files/4041851/ik.cpp.zip)

dudasdavid, 2020-11-24:
@JaehyunShim, I created an IKFast plugin based on @nxdefiant's great work:
https://github.com/dudasdavid/open_manipulator_ikfast_plugin

But I still have some difficulties with the gripper joint, see my video:
https://youtu.be/zHTLPNMEGCs

It's a pain in the back to generate such a plugin with an - let's say - obsolete and under-documented toolchain, and I still have no clue what is the problem with the plugin.

As I can see on your Facebook page (https://www.facebook.com/robotis.company/videos/2198266990449746/), you have a working plugin for MoveIt. Could you please share your plugin with the community?

I think this issue should be re-opened because it's not moveit's responsibility to supply such a plugin for a 1000+ EUR kit. Especially if it's advertised that it works with MoveIt. I'm happy to open a new issue if it works better for you.

jeongnow97, 2022-06-23:
@nxdefiant
@JaehyunShim 

Hello, What should I do to use that IKFast plugin?

I encounterd the same problem that I can't control open-manipulator's end-effector

I already installed all package related to open-manipulator and move-it

Thank you :)

dudasdavid, 2022-06-23:
@jeongnow97 unfortunately there is still no official package for that. You can download my plugin above and build the workspace.

After that you can modify the following config: open_manipulator_controls/open_manipulator_moveit_config/config/kinematics.yaml:

Original:

arm:
  kinematics_solver: kdl_kinematics_plugin/KDLKinematicsPlugin
  kinematics_solver_search_resolution: 0.005
  kinematics_solver_timeout: 0.005

Modified:

arm:
  kinematics_solver: open_manipulator/IKFastKinematicsPlugin
  kinematics_solver_search_resolution: 0.005
  kinematics_solver_timeout: 0.005

Of course you can do the modification with MoveIt assistant if you prefer.

jeongnow97, 2022-06-27:
Thank you!

I have one more question

Download that package in catkin_ws/src right?

It doesn't work..

I executed demo.launch

Error said
[ERROR] [1656305223.759995857]: The kinematics plugin (arm) failed to load. Error: According to the loaded plugin descriptions the class open_manipulator/IKFastKinematicsPlugin with base class type kinematics::KinematicsBase does not exist. Declared types are  cached_ik_kinematics_plugin/CachedKDLKinematicsPlugin cached_ik_kinematics_plugin/CachedSrvKinematicsPlugin kdl_kinematics_plugin/KDLKinematicsPlugin lma_kinematics_plugin/LMAKinematicsPlugin moveit_opw_kinematics_plugin/MoveItOPWKinematicsPlugin pr2_arm_kinematics/PR2ArmKinematicsPlugin prbt_manipulator/IKFastKinematicsPlugin srv_kinematics_plugin/SrvKinematicsPlugin trac_ik_kinematics_plugin/TRAC_IKKinematicsPlugin
[ERROR] [1656305223.760018732]: Kinematics solver could not be instantiated for joint group arm.

Thank you :)

nxdefiant, 2022-06-27:
Did you run catkin_make before demo.launch?

jeongnow97, 2022-06-27:
Of course

Well mine was catkin build..

But there's no difference right?

jeongnow97, 2022-06-27:
It's work

Reboot the computer and work now!

Thank you so much

JaehyunShim, 2022-06-27:
I left ROBOTIS last year and someone else should be in charge of the maintenance role as of last summer. Please open a new thread if you have any further questions.