ROBOTIS-GIT/open_manipulator #144: Replace Dynamixel Joint 12 in OpenManipulator

shyne98, 2020-01-28:
Hi,

I have an OpenManipulator arm attached to a Turtlebot3 Waffle Pi. One of the OpenManipulator motors, Joint 12, needed to be replaced because the gears were damaged. 

I ordered a new motor, Dynamixel XM430-W350-T, and installed it in the Joint 12 position on the OpenManipulator arm.  

When reloading the firmware using the Shell Script (http://emanual.robotis.com/docs/en/platform/turtlebot3/manipulation/#shell-script), the firmware loads successfully and I receive the "jump_to_fw" code. However, Joint 12 isn't accepting the firmware, see the following video for details: https://youtu.be/id97lRlEP4U. 

Any thoughts as to what the problem could be? 

[om_with_tb3_fw_scripts.txt](https://github.com/ROBOTIS-GIT/open_manipulator/files/4125031/om_with_tb3_fw_scripts.txt)

rjshim, 2020-01-29:
@shyne98 

Did you set up the motor with [Dynamixel Workbench](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/) before using it? Check if the motor number and baud rate are set to 12 and 1M respectively. 

Ryan

rjshim, 2020-02-24:
As there has been no update, I will close this issue. Ryan

closed 2020-02-24 by rjshim