ROBOTIS-GIT/open_manipulator #149: Any issue if I use the XM430W350R or XH430-W350R to make the OpenManipulator-X and its friends?

robotiscustomer, 2020-02-25:
Hi, if I use the XM430-W350R and/or XH430-W350R to make the OpenManipulator-X (RM-X52-TNM) and its friends such as the 6-DOF SARA, SCARA, Stewart, LINK, DELTA, PLANAR, LINEAR, etc. will the robots, sample programs and the corresponding gazebo simulations all work without any hardware and software modification? Will the simulated behaviors still match those of the real robots driven by the XM/H-430W305R actuators?

In the e-manual, it states that TTL is used as the Communication protocol for the OpenManipulator-X and its friends. Will I have problem if I use the RS-485 dynamixels instead?

I read that some users have used the XL-320 motors alone to make the OpenManipulator. Besides the low payload and not being able to use current control method to control the gripper, are there any other issues?

rjshim, 2020-03-04:
@robotiscustomer 

There will be no need for change in source code since XM and XH motors use the same control table. Test it and let us know if you have any other questions.

Ryan

closed 2020-03-04 by rjshim

robotiscustomer, 2020-03-04:
@rjshim 

Thanks.  I have not used the X-series nor ROS before. I am trying to decide which model to use. I plan to buy 2 or 3 motors for testing first and see how it goes. If I build the OpenManipulator-X with less than 5 motors, will the sample programs still work? In the testing stage, I don't need the gripper nor the wrist parts.

rjshim, 2020-03-04:
@robotiscustomer I recommend you buy [the whole kit](http://en.robotis.com/model/page.php?co_id=prd_openmanipulator) if you are a beginner user. The answer is no. It wouldn't work without five motors connected.