ROBOTIS-GIT/open_manipulator #153: MoveIt option

jonny214, 2020-03-11:
Hello,

I would like to have moveit option code.
http://emanual.robotis.com/docs/en/platform/openmanipulator_x/ros_operation/#ros-operation

This option was removed last week.
Will you plan to update it?
If no, I will find previous code from history.

Thanks,
Jun

rjshim, 2020-03-13:
Hi, 

The previous controller + moveit was implemented in the wrong way so I removed it. You can use the previous code as you mentioned in your comment if you need to even though it is not the right way of using hardware_interface + move_group. We currently do not have any concrete plans for the new update yet. 

Ryan

rjshim, 2020-03-16:
Refer to this link to use MoveGroup + JointTrajectoryController.

Ryan

jonny214, 2020-03-16:
Hi Ryan,

Thank you so much for kindly comment.
And, I understood the situation.

I would refer to previous code and your "link".

Thanks a lot,
Jun

closed 2020-03-16 by jonny214