ROBOTIS-GIT/open_manipulator #155: Which ROS1 Setup to follow if I have both OM and Turtlebot 3?

robotiscustomer, 2020-03-20:
Hi, I have installed Ubuntu 16.04. I plan to use the OpenManipulator-X and TurtleBot3 separately sometimes but combine them together other times. 

In the Open Manipulator-X emanual under [ROS] Setup, there are:

5.2 Install ROS on PC
5.3 Install ROs Packages
5.4 Communication Converter
      - U2D2
      - OpenCR

In the TurtleBot3 e-Manual, there are:

6.1 PC Setup
6.3 OpenCR Setup
      -- OpenCR Firmware Upload for TB3
13. [ROS1] Manipulation

In my case, which one should I follow? Since OpenManipulator on a stationary Turtlebot3 is like it is on the Base Plate, I would guess it is better to follow the steps in the TurtleBot3 e-Manual. However, some of the packages described in the OP-X emanual are different from those listed in the [ROS1] Manipulation section. If I follow all the steps in both emanual, will the system get messed up? 

What is OpenCR Firmware Upload for TB3 for? If I upload it, will it limit the usage of the OpenCR board to Turtlebot3 only? Will I still be able to use the OpenCR board to control other robots besides the TB3?

ROBOTIS-Will, 2020-04-28:
@robotiscustomer 
Hi, sorry for the late reply.
Since TurtleBot3 is a main product, please use TurtleBot3 instructions in order to use OpenMANIPULATOR-X on it.
[http://emanual.robotis.com/docs/en/platform/turtlebot3/manipulation/#manipulation](http://emanual.robotis.com/docs/en/platform/turtlebot3/manipulation/#manipulation)

There is a specific OpenCR firmware for TurtleBot3 + OpenMANIPULATOR-X (turtlebot3_with_open_manipulator_core.ino)
Only one firmware can be installed to OpenCR at a time.

Thank you.

ROBOTIS-Will, 2021-03-05:
Closing this issue as there isn't any recent activity. Please feel free to reopen when necessary.

closed 2021-03-05 by ROBOTIS-Will