ROBOTIS-GIT/open_manipulator #156: Broken link in emanual for Open Manipulator-X

robotiscustomer, 2020-03-20:
Hi, under Section 5.3.2 OpenCR, it states that "if you want to operate OpenMANIPULATOR-X on embedded system (OpenCR) without ROS, please refer OpenCR Setup."

The OpenCR Setup link is broken.

Please fix it.

ROBOTIS-David, 2020-04-12:
Thank you for your report, the broken link will be updated from the next e-Manual update.

robotiscustomer, 2020-04-23:
Thanks David. In regard to the e-Manual, what is the best way to print it? If I just print as is, sometimes 1/3-1/2 of the pages are for Table of Contents. Some codes/commands are cut off and not being printed. If I hit Edit on GiHub and print, all the section numbers are missing. Also, how can I tell which version of the e-Manual I am reading and the version of the manual posted on the website?

ROBOTIS-David, 2020-04-23:
Hi, Thanks for your reply, 

- If you are using Chrome, just save as **_webpage,complete_**, which preserves most contents on the page, and then print the page. 
- Unfortunately, the current e-Manual does not provide the version information as the manual is often upadated for new features. 

I hope this information is helpful for you .

Many thanks, 

David Park.

robotiscustomer, 2020-04-24:
Thanks David. Please keep safe and healthy.

ROBOTIS-David, 2020-04-24:
No worries, If you have any further questions requiring assistance, please feel free to reach out to us at any time. But if not, please close this thread.

robotiscustomer, 2020-04-27:
Thanks. It could be helpful if the date of the last updated information is included.

closed 2020-04-28 by robotiscustomer