ROBOTIS-GIT/open_manipulator #159: Error: package 'open_manipulator_controller' not found

robotiscustomer, 2020-04-09:
Hi,

I am still having issues in Section 6.1 Launch Controller. Yesterday, I was unable to copy and paste the output of my screen to the Robotis forum due to the internal server security of your company. CS suggested me to move to Github. Hope you could help to get my issues resolved. Thank you.

As a recap, here are the things that we have done so far. Please note that there were issues with installing ROS using the method presented in the OpenManipulator e-Manual. I used the procedure described on the official ROS site to install ROS1 under Ubuntu 16.04 on a linux virtual machine. Currently my computer is not connected to the OpenManipulator-X yet.

http://en.robotis.com/service/forum_view.php?slg=&page_type=&bbs_no=2407806

http://en.robotis.com/service/forum_view.php?bbs_no=2407801&page=1&save_sca=&sca=&save_stx=&stx=&sfl=

http://en.robotis.com/service/forum_view.php?bbs_no=2407774&page=4&save_sca=&sca=&save_stx=&stx=&sfl=

---------------------------------------------------------------------------------------------
I have copied and pasted the output of my screen as requested.

user@ubuntu:~/catkin_ws$ ls
build  devel  src
user@ubuntu:~/catkin_ws$ rm -rf src
user@ubuntu:~/catkin_ws$ cd ~
user@ubuntu:~$ sudo apt-get install ros-kinetic-ros-controllers ros-kinetic-gazebo* ros-kinetic-moveit* ros-kinetic-industrial-core
[sudo] password for user: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'ros-kinetic-gazebo-ros' for glob 'ros-kinetic-gazebo*'
Note, selecting 'ros-kinetic-gazebo-msgs' for glob 'ros-kinetic-gazebo*'
Note, selecting 'ros-kinetic-gazebo-ros-control' for glob 'ros-kinetic-gazebo*'
Note, selecting 'ros-kinetic-gazebo-dev' for glob 'ros-kinetic-gazebo*'
Note, selecting 'ros-kinetic-gazebo-plugins' for glob 'ros-kinetic-gazebo*'
Note, selecting 'ros-kinetic-gazebo-ros-pkgs' for glob 'ros-kinetic-gazebo*'
Note, selecting 'ros-kinetic-moveit-fake-controller-manager' for glob 'ros-kinetic-moveit*'
Note, selecting 'ros-kinetic-moveit-visual-tools' for glob 'ros-kinetic-moveit*'
Note, selecting 'ros-kinetic-moveit-msgs' for glob 'ros-kinetic-moveit*'
Note, selecting 'ros-kinetic-moveit-pr2' for glob 'ros-kinetic-moveit*'
Note, selecting 'ros-kinetic-moveit-ros' for glob 'ros-kinetic-moveit*'
Note, selecting 'ros-kinetic-moveit-runtime' for glob 'ros-kinetic-moveit*'
Note, selecting 'ros-kinetic-moveit-config-m1013' for glob 'ros-kinetic-moveit*'
Note, selecting 'ros-kinetic-moveit-sim-controller' for glob 'ros-kinetic-moveit*'
Note, selecting 'ros-kinetic-moveit-ros-robot-interaction' for glob 'ros-kinetic-moveit*'
Note, selecting 'ros-kinetic-moveit-ros-planning-interface' for glob 'ros-kinetic-moveit*'
Note, selecting 'ros-kinetic-moveit-config-m1509' for glob 'ros-kinetic-moveit*'
Note, selecting 'ros-kinetic-moveit-experimental' for glob 'ros-kinetic-moveit*'
Note, selecting 'ros-kinetic-moveit-ros-control-interface' for glob 'ros-kinetic-moveit*'
Note, selecting 'ros-kinetic-moveit-planners-chomp' for glob 'ros-kinetic-moveit*'
Note, selecting 'ros-kinetic-moveit-commander' for glob 'ros-kinetic-moveit*'
Note, selecting 'ros-kinetic-moveit-kinematics' for glob 'ros-kinetic-moveit*'
Note, selecting 'ros-kinetic-moveit-controller-manager-example' for glob 'ros-kinetic-moveit*'
Note, selecting 'ros-kinetic-moveit-ros-move-group' for glob 'ros-kinetic-moveit*'
Note, selecting 'ros-kinetic-moveit-ros-perception' for glob 'ros-kinetic-moveit*'
Note, selecting 'ros-kinetic-moveit' for glob 'ros-kinetic-moveit*'
Note, selecting 'ros-kinetic-moveit-chomp-optimizer-adapter' for glob 'ros-kinetic-moveit*'
Note, selecting 'ros-kinetic-moveit-setup-assistant' for glob 'ros-kinetic-moveit*'
Note, selecting 'ros-kinetic-moveit-simple-controller-manager' for glob 'ros-kinetic-moveit*'
Note, selecting 'ros-kinetic-moveit-config-m0609' for glob 'ros-kinetic-moveit*'
Note, selecting 'ros-kinetic-moveit-config-m0617' for glob 'ros-kinetic-moveit*'
Note, selecting 'ros-kinetic-moveit-ros-warehouse' for glob 'ros-kinetic-moveit*'
Note, selecting 'ros-kinetic-moveit-python' for glob 'ros-kinetic-moveit*'
Note, selecting 'ros-kinetic-moveit-resources' for glob 'ros-kinetic-moveit*'
Note, selecting 'ros-kinetic-moveit-plugins' for glob 'ros-kinetic-moveit*'
Note, selecting 'ros-kinetic-moveit-ros-manipulation' for glob 'ros-kinetic-moveit*'
Note, selecting 'ros-kinetic-moveit-ros-planning' for glob 'ros-kinetic-moveit*'
Note, selecting 'ros-kinetic-moveit-planners-ompl' for glob 'ros-kinetic-moveit*'
Note, selecting 'ros-kinetic-moveit-planners' for glob 'ros-kinetic-moveit*'
Note, selecting 'ros-kinetic-moveit-ros-visualization' for glob 'ros-kinetic-moveit*'
Note, selecting 'ros-kinetic-moveit-core' for glob 'ros-kinetic-moveit*'
Note, selecting 'ros-kinetic-moveit-ros-benchmarks' for glob 'ros-kinetic-moveit*'
ros-kinetic-gazebo-dev is already the newest version (2.5.19-1xenial-20191214-003634+0000).
ros-kinetic-gazebo-msgs is already the newest version (2.5.19-1xenial-20191214-085403+0000).
ros-kinetic-gazebo-plugins is already the newest version (2.5.19-1xenial-20200109-025835+0000).
ros-kinetic-gazebo-ros is already the newest version (2.5.19-1xenial-20200109-034212+0000).
[... 609 lines cut ...]
setting /run_id to 7dce0586-7ab7-11ea-b0d0-001c426a2b68
process[rosout-1]: started with pid [20829]
started core service [/rosout]

###################################################################
Went back to the original terminal and typed:
[100%] Built target open_manipulator_controller
user@ubuntu:~/catkin_ws$ rosrun open_manipulator_controller create_udev_rules
[rospack] Error: package 'open_manipulator_controller' not found

###################################################################
Also tried Section 6.1 Launch Controller

user@ubuntu:~/catkin_ws$ roslaunch open_manipulator_controller open_manipulator_controller.launch
[open_manipulator_controller.launch] is neither a launch file in package [open_manipulator_controller] nor is [open_manipulator_controller] a launch file name
The traceback for the exception was written to the log file
#####################################################################
Executing source ~/.bashrc does not help.

user@ubuntu:~/catkin_ws$ source ~/.bashrc
user@ubuntu:~/catkin_ws$ rosrun open_manipulator_controller create_udev_rules
[rospack] Error: package 'open_manipulator_controller' not found
user@ubuntu:~/catkin_ws$ roslaunch open_manipulator_controller open_manipulator_controller.launch
[open_manipulator_controller.launch] is neither a launch file in package [open_manipulator_controller] nor is [open_manipulator_controller] a launch file name
The traceback for the exception was written to the log file

robotiscustomer, 2020-04-09:
Sorry not sure why some lines got crossed out by Github.

ROBOTIS-Will, 2020-04-29:
@robotiscustomer 
Looks like you're missing some libraries or those are not properly installed.
Please refer to below to see if this is a similar case with yours.
https://www.reddit.com/r/Ubuntu/comments/g1sh09/the_imported_target_qt5gui_references_the_file/

ROBOTIS-Will, 2021-03-05:
Closing this issue as there isn't any recent activity. Please feel free to reopen when necessary.

closed 2021-03-05 by ROBOTIS-Will

LOKAKSHAI, 2021-05-17:
When I'm trying to build the system it show the error

Invoking "make cmake_check_build_system" failed

ROBOTIS-Will, 2021-05-17:
@LOKAKSHAI 
Which Ubuntu and ROS do you use?
Could you copy and paste your terminal output?

LOKAKSHAI, 2021-05-18:
Hi team,

Thanks for your reply.

I'm using
 ubuntu 18.04
 Ros 1

Kindly check and let me know about your feedback.

ROBOTIS-Will, 2021-05-24:
@LOKAKSHAI 
Have you run the `catkin_make` command under the `catkin_ws` directory?
Since this is not the issue of open manipulator, please refer to ROS Answers for more information.
There are many questions with the same error message.
https://answers.ros.org/questions/query:Invoking%20%22make%20cmake_check_build_system%22%20failed/
Thanks