ROBOTIS-GIT/open_manipulator #163: Error of “Failed to open the port”

EndlessLoops, 2020-04-14:
**Test Background**

TX2（JP4.2.3）+ Ubuntu 18.04 + ROS  Melodic + OpenCR + Open_Manipulator

**The problem I encountered**

I found when When the robot's USB cable was connected to TX2 for the first time, it would always give the error of “Failed to open the port”. But after re-executing the program, it will work normally.
And it works fine on my PC and TX2（JP3.0）.

**Error Code**

```
ubuntu@TX2:~/openManipulator_ws/src/open_manipulator/open_manipulator_controller/launch$ roslaunch open_manipulator_controller open_manipulator_controller.launch
... logging to /home/ubuntu/.ros/log/51ef0bba-7dfd-11ea-bb59-00044b8c6419/roslaunch-TX2-9933.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
started roslaunch server http://localhost:43255/
SUMMARY
========
PARAMETERS
* /open_manipulator/control_period: 0.01
* /open_manipulator/moveit_sample_duration: 0.05
* /open_manipulator/planning_group_name: arm
* /open_manipulator/using_moveit: False
* /open_manipulator/using_platform: True
* /rosdistro: melodic
* /rosversion: 1.14.3
NODES
  /
    open_manipulator (open_manipulator_controller/open_manipulator_controller)
auto-starting new master
process[master]: started with pid [9947]
ROS_MASTER_URI=http://localhost:11311
setting /run_id to 51ef0bba-7dfd-11ea-bb59-00044b8c6419
process[rosout-1]: started with pid [9958]
started core service [/rosout]
process[open_manipulator-2]: started with pid [9962]
[PortHandlerLinux::SetupPort] Error opening serial port!
[ERROR] [DynamixelDriver] Failed to open the port!
[ERROR] [TxRxResult] Failed transmit instruction packet!
[ERROR] Please check your Dynamixel ID
[ERROR] [TxRxResult] Failed transmit instruction packet!
[ERROR] Please check your Dynamixel ID
[ERROR] [TxRxResult] Failed transmit instruction packet!
[ERROR] Please check your Dynamixel ID
[ERROR] [TxRxResult] Failed transmit instruction packet!
[ERROR] Please check your Dynamixel ID
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[PortHandlerLinux::SetupPort] Error opening serial port!
[ERROR] [DynamixelDriver] Failed to open the port!
[ERROR] [TxRxResult] Failed transmit instruction packet!
[ERROR] Please check your Dynamixel ID
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelWorkbench] Failed to change torque status!
[ERROR] [DynamixelWorkbench] Failed to change torque status!
[ERROR] [DynamixelWorkbench] Failed to change torque status!
[ERROR] [DynamixelWorkbench] Failed to change torque status!
[ERROR] [DynamixelWorkbench] Failed to change torque status!
[ERROR] [TxRxResult] Failed transmit instruction packet!
[ERROR] groupSyncRead getdata failed
[ERROR] groupSyncRead getdata failed
[ERROR] groupSyncRead getdata failed
[open_manipulator-2] process has died [pid 9962, exit code -11, cmd /home/ubuntu/openManipulator_ws/devel/lib/open_manipulator_controller/open_manipulator_controller /dev/ttyACM0 1000000 __name:=open_manipulator __log:=/home/ubuntu/.ros/log/51ef0bba-7dfd-11ea-bb59-00044b8c6419/open_manipulator-2.log].
log file: /home/ubuntu/.ros/log/51ef0bba-7dfd-11ea-bb59-00044b8c6419/open_manipulator-2*.log
^C[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

EndlessLoops, 2020-04-20:
Have you ever tested on the Ubuntu 1804？
I add a virtual machine with 1804 and I found it also was enstable in opening serial port .
It can work normally in virtual machine with 1604

ROBOTIS-Will, 2020-04-29:
@EndlessLoops 
Hi, unfortunately we do not have TX2 so won't be able to provide accurate information. By default, the launch file will search for the /dev/ttyACM0. Make sure that the TX2 is assigning ttyACM0 to the OpenCR or else you should modify the `open_manipulator_controller.launch` file to connect proper port.

ROBOTIS-Will, 2021-03-05:
Closing this issue as there isn't any recent activity. Please feel free to reopen when necessary.

closed 2021-03-05 by ROBOTIS-Will