ROBOTIS-GIT/open_manipulator #165: error: could not convert ...... from ‘void’ to ‘bool’

phillong1, 2020-04-17:
In member function ‘bool open_manipulator_controller::OpenManipulatorController::goalJointSpacePathCallback(open_manipulator_msgs::SetJointPosition::Request&, open_manipulator_msgs::SetJointPosition::Response&)’:
/home/flong/simulation_ws/src/open_manipulator/open_manipulator_controller/src/open_manipulator_controller.cpp:210:45: error: could not convert ‘((open_manipulator_controller::OpenManipulatorController*)this)->open_manipulator_controller::OpenManipulatorController::open_manipulator_.OpenManipulator::<anonymous>.robotis_manipulator::RobotisManipulator::makeJointTrajectory(std::vector<double>(target_angle), req.open_manipulator_msgs::SetJointPositionRequest_<std::allocator<void> >::path_time, std::vector<robotis_manipulator::_Point>())’ from ‘void’ to ‘bool’
   if (!open_manipulator_.makeJointTrajectory(target_angle, req.path_time))
********************************************************************************************
I looked at your code, 'makeJointTrajectory' are all void type......

  void makeJointTrajectory(std::vector<double> goal_joint_position, double move_time, std::vector<JointValue> present_joint_value = {});
  void makeJointTrajectory(std::vector<JointValue> goal_joint_value, double move_time, std::vector<JointValue> present_joint_value = {});
  void makeJointTrajectory(Name tool_name, Eigen::Vector3d goal_position, double move_time, std::vector<JointValue> present_joint_value = {});
  void makeJointTrajectory(Name tool_name, Eigen::Matrix3d goal_orientation, double move_time, std::vector<JointValue> present_joint_value = {});
  void makeJointTrajectory(Name tool_name, KinematicPose goal_pose, double move_time, std::vector<JointValue> present_joint_value = {});

ROBOTIS-Will, 2020-04-29:
@phillong1 
Hi, the makeJointTrajectory function in `open_manipulator_controller.cpp` comes from the `robotis_manipulator` package and they are all [defined with boolean returns in the latest code](https://github.com/ROBOTIS-GIT/robotis_manipulator/blob/master/src/robotis_manipulator/robotis_manipulator.cpp#L919).
If you have installed the package with binary source such as `apt install ros-kinetic...` please download the source code and build from it as [instructed in the eManual](http://emanual.robotis.com/docs/en/platform/openmanipulator_x/ros_setup/#install-ros-packages).
We're preparing to update the binary packages soon.
Thank you.

ROBOTIS-Will, 2021-03-05:
Closing this issue as there isn't any recent activity. Please feel free to reopen when necessary.

closed 2021-03-05 by ROBOTIS-Will