ROBOTIS-GIT/open_manipulator #173: issues in catkin_make

albinbiju, 2020-05-09:


ROBOTIS-Will, 2020-05-12:
@albinbiju 
open_manipulator package has dependencies in several other packages.
Please refer to below eManual section for necessary packages.
http://emanual.robotis.com/docs/en/platform/openmanipulator_x/ros_setup/#install-ros-packages

albinbiju, 2020-05-13:
But I still don't get the joints topic published in the latest versions of the open manipulator

ROBOTIS-Will, 2020-05-13:
@albinbiju 
In the recent update, there was a change in major controller interface and now it is designed to use moveit. You can get each joint value from the position in /joint_states topic.
[image]

ROBOTIS-Will, 2021-03-05:
Closing this issue as there isn't any recent activity. Please feel free to reopen when necessary.

closed 2021-03-05 by ROBOTIS-Will