ROBOTIS-GIT/open_manipulator #174: OpenMANIPULATOR not listed in Topic Monitor

robotiscustomer, 2020-05-11:
Hi, I posted this issue on Robotis Q&A Forum last month but it is still not resolved. Can you help please?

http://en.robotis.com/service/forum_view.php?bbs_no=2407823&page=3&save_sca=&sca=&save_stx=&stx=&sfl=

ROBOTIS-Will, 2020-05-12:
Hi,
[This eManual instruction](http://emanual.robotis.com/docs/en/platform/openmanipulator_x/ros_operation/) is specifically written only for the OpenMANIPULATOR-X. If you attached the OpenMANIPULATOR-X on top of TurtleBot3, the instruction may not work for your configuration.
Please tell us more details such as how you configured your hardware, and which commands you entered so that we can try to reproduce your issue and help to resolve it.
Unfortunately, there are too many ways to make things wrong and we cannot check all of them.
Some of the images doesn't seem updated so I'll request to update these images and contents, but this shouldn't be a problem for running the instruction.

robotiscustomer, 2020-05-12:
Thanks Will. The OpenManipulator-X is on its own. I have not put it on the Turtlebot3 yet. I just followed the instructions on the e-Manual and whenever I encountered an issue, I posted here or the Q&A forum.

ROBOTIS-Will, 2020-05-13:
Below is the image you posted that shows multiple camera related topics. 
[image]
If you correctly followed the manual instruction and launched only one launch file `$ roslaunch open_manipulator_controller open_manipulator_controller.launch` you will see about 5 topics only.

Just to make sure that you have the latest source code, please run below command in your terminal and relaunch the launch file.
Below command basically iterates each source code repository and set it to master branch then update to the latest.
Then, finally build the source code to make them effective.
```
$ cd ~/catkin_ws/src/open_manipulator && git checkout master && git pull 
$ cd ~/catkin_ws/src/open_manipulator_msgs && git checkout master && git pull 
$ cd ~/catkin_ws/src/open_manipulator_simulations && git checkout master && git pull 
$ cd ~/catkin_ws/src/robotis_manipulator && git checkout master && git pull 
$ cd ~/catkin_ws && catkin_make
```

ROBOTIS-Will, 2021-03-05:
Closing this issue as there isn't any recent activity. Please feel free to reopen when necessary.

closed 2021-03-05 by ROBOTIS-Will