ROBOTIS-GIT/open_manipulator #176: The difference between the two control modes？

Sunchanghao, 2020-06-01:
one is :
    roslaunch open_manipulator_gazebo open_manipulator_gazebo.launch
    roslaunch open_manipulator_controller open_manipulator_controller.launch
another is:
     roslaunch open_manipulator_controllers joint_trajectory_controller.launch sim:=false

i find that these two modes to control manipulator is different, when i need use service(such as SetJointPosition) to control manipulator , only the first method could make the arm move.
i want to know why, and what the difference between these two methods.
thx～

ROBOTIS-Will, 2020-06-08:
Hi,
roslaunch open_manipulator_controller openc_manipulator_controller.launch command uses the ROBOTIS manipulator library that defines a custom service `SetJointPosition` in it while the other uses the position_controllers in the ros-controls package, therefore, the latter does not accept SetJointPosition service.

ROBOTIS-Will, 2021-03-05:
Closing this issue as there isn't any recent activity. Please feel free to reopen when necessary.

closed 2021-03-05 by ROBOTIS-Will