ROBOTIS-GIT/open_manipulator #177: Open Manipulator Controller does not  come up when using OpenCR

edcepp, 2020-06-12:
### Environment
Ubuntu 18.04, OpenCR running usb_to_dxl

### Test 
```
$ export RCUTILS_CONSOLE_STDOUT_LINE_BUFFERED=1

$ ros2 run open_manipulator_x_controller open_manipulator_x_controller 
port_name and baud_rate are set to /dev/ttyUSB0, 1000000 
[INFO] [open_manipulator_x_controller]: Ready to Simulate OpenManipulator-X on Gazebo
```
The OpenCR is on /dev/ttyACM0. So I entered

```
$ ros2 run open_manipulator_x_controller open_manipulator_x_controller /dev/ttyACM0 1000000
port_name and baud_rate are set to /dev/ttyACM0, 1000000 
[INFO] [open_manipulator_x_controller]: Ready to Simulate OpenManipulator-X on Gazebo

```
### Problem
The servos do not resist being turned. Teleop keyboard does not control the servos.

### Question 1
How do I start up the Open Manipulator Controller?

### Setup Verification

- Dynamixel Wizard2 runs fine under OpenCR and finds all 5 servos during its scan
- Teleop keyboard runs properly under U2D2. It controls all 5 servos.

### Question 2
Once I get the controller going how do I start up teleop keyboard? This is what I tried.

```
$ export RCUTILS_CONSOLE_STDOUT_LINE_BUFFERED=1

$ ros2 run open_manipulator_x_teleop open_manipulator_x_teleop_keyboard
```
and
```
$ ros2 run open_manipulator_x_teleop open_manipulator_x_teleop_keyboard /dev/ttyACM0 1000000
```

ROBOTIS-Will, 2020-06-29:
@edcepp 
Thank you for reporting the issue.
Please make sure that you are not running with simulation settings; for example, in the `open_manipulator_x_controller.cpp`file, line 72 defines simulation(gazebo) init. This is by default set to `false`.
```
this->get_parameter_or<bool>("sim", sim_, false);
```

I've revised the [eManual instructions](https://emanual.robotis.com/docs/en/platform/openmanipulator_x/ros2_controller_package/#launch-controller) so that it can properly be launched.
Please see below test result that I ran on my Linux Mint(Ubuntu 18.04 base), ROS2 Dashing with the most up to date github source codes.
[image]
[image]
[image]
[image]

We are currently working on reviewing and modifying OpenMANIPULATOR-X and TurtleBot3 source code and eManual contents, and hope we can make our documentations easier to follow.
Always appreciate your feedback and error reports.

edcepp, 2020-06-30:
Thanks you. You gave me the clue I needed. This is what I did:
 
Retest
$ ros2 run open_manipulator_x_controller open_manipulator_x_controller /dev/ttyACM0 1000000
port_name and baud_rate are set to /dev/ttyACM0, 1000000 
[INFO] [open_manipulator_x_controller]: Ready to Simulate OpenManipulator-X on Gazebo
 
The Gazebo reference tells me I’m in simulation (not hardware) mode. I check the parameters.
$ ros2 param list
/open_manipulator_x_controller:
  control_period
  use_platform
  use_sim_time
 
I check to see what the value of the use_platform parameter should be. Notice the parameter name is not “sim”
$ cat src/open_manipulator/open_manipulator_x_controller/param/open_manipulator_x_controller_params.yaml
open_manipulator_x_controller:
  ros__parameters:
    use_platform: true
    control_period: 0.010
 
I check to see what the value of the use_platform parameter is.
$ ros2 param get /open_manipulator_x_controller use_platform
Parameter not set.
 
This is a problem. There is something wrong with parameters. I edit the code to force the use_platform to true.
$ diff open_manipulator_x_controller.cpp.org open_manipulator_x_controller.cpp
72c72
<  this->get_parameter_or<bool>("use_platform", use_platform_, false);
---

>  this->get_parameter_or<bool>("use_platform", use_platform_, true);
 
I rebuild
$ colcon build --symlink-install
 
Rerun
$ ros2 run open_manipulator_x_controller open_manipulator_x_controller /dev/ttyACM0 1000000
port_name and baud_rate are set to /dev/ttyACM0, 1000000 
Joint Dynamixel ID : 11, Model Name : XM430-W350
Joint Dynamixel ID : 12, Model Name : XM430-W350
Joint Dynamixel ID : 13, Model Name : XL430-W250
Joint Dynamixel ID : 14, Model Name : XL430-W250
Gripper Dynamixel ID : 15, Model Name :XL430-W250
[ERROR] [DynamixelWorkbench] Failed to set Current Based Position Control Mode!
[INFO] [open_manipulator_x_controller]: Succeeded to Initialise OpenManipulator-X Controller
 
It starts up as expected. (The ERROR is because I installed an XL servo and not an XM. This is not a fatal error.)
 
To test I run keyboard teleop in terminal 2. It works as expected. I am able to control the manipulator controlled with an OpenCR.
$ export RCUTILS_CONSOLE_STDOUT_LINE_BUFFERED=1
 
$ ros2 run open_manipulator_x_teleop open_manipulator_x_teleop_keyboard

ROBOTIS-Will, 2020-07-01:
@edcepp 
Thank you for sharing your result.
The `use_platform` parameter had been renamed to `sim` in the most recent source code, but you made the correct modification.
We are currently discussing our manual format that will be easier to follow for the first time users.
I hope it can be distributed within a couple weeks.
Thank you very much for your valuable feedback.

ROBOTIS-Will, 2021-03-05:
Closing this issue as there isn't any recent activity. Please feel free to reopen when necessary.

closed 2021-03-05 by ROBOTIS-Will