ROBOTIS-GIT/open_manipulator #179: [ERROR] [OpenManipulator Chain Custom]fail to solve inverse kinematics

EndlessLoops, 2020-07-08:
Use_platform: OpenManipulator + OpenCR + Intel D435 i+ TX2 + ubuntu16.04

I  tested the pick and place demo . I got an error when program executed to the statement [   kinematics_position.push_back(ar_marker_pose.at(i).position[0]);]( https://github.com/ROBOTIS-GIT/open_manipulator_perceptions/blob/eebbbd4eef28f610df027d3b3c5c1f81a41a380c/open_manipulator_pick_and_place/src/open_manipulator_pick_and_place.cpp#L243).

Program can't finsh the pick action,and show the error tip

```
~$ roslaunch open_manipulator_controller open_manipulator_controller.launch 
... logging to /home/ubuntu/.ros/log/cca307de-c101-11ea-a659-00044b8d772a/roslaunch-ROS-15870.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:45921/

SUMMARY
========

PARAMETERS
 * /open_manipulator_controller/control_period: 0.01
 * /open_manipulator_controller/using_platform: True
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    open_manipulator_controller (open_manipulator_controller/open_manipulator_controller)

auto-starting new master
process[master]: started with pid [15880]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to cca307de-c101-11ea-a659-00044b8d772a
process[rosout-1]: started with pid [15893]
started core service [/rosout]
process[open_manipulator_controller-2]: started with pid [15908]
port_name and baud_rate are set to /dev/ttyACM0, 1000000 
Joint Dynamixel ID : 11, Model Name : XM430-W350
Joint Dynamixel ID : 12, Model Name : XM430-W350
Joint Dynamixel ID : 13, Model Name : XM430-W350
Joint Dynamixel ID : 14, Model Name : XM430-W350
Gripper Dynamixel ID : 15, Model Name :XM430-W350
[INFO] Succeeded to init /open_manipulator_controller
[ERROR] [OpenManipulator Chain Custom]fail to solve inverse kinematics
[ERROR] [TASK_TRAJECTORY] Fail to solve IK
[ERROR] [OpenManipulator Chain Custom]fail to solve inverse kinematics
[ERROR] [TASK_TRAJECTORY] Fail to solve IK
[ERROR] [TxRxResult] There is no status packet!
[ERROR] groupSyncRead getdata failed
[ERROR] [TxRxResult] There is no status packet!
[ERROR] groupSyncRead getdata failed
[ERROR] [TxRxResult] Incorrect status packet!
[ERROR] groupSyncRead getdata failed
[ERROR] groupSyncRead getdata failed
[ERROR] groupSyncRead getdata failed
[ERROR] [OpenManipulator Chain Custom]fail to solve inverse kinematics
[ERROR] [TASK_TRAJECTORY] Fail to solve IK
```

What causes this situation in general, and how should I do？

I don't know much about robot arm，and looking forward to your reply.

EndlessLoops, 2020-07-13:
I found if the X value of position exceed 2.0 ,the error message  "fail to solve inverse kinematics"  will appear in terminal of open_manipulator_controller ｗhether on tx2 or PC .The arm can't move normally to the corresponding position on TX2 but it can move normally on PC.

ROBOTIS-Will, 2021-07-01:
Hi @EndlessLoops 
I sincerely apologize about super delayed response on this. While I was skimming through the remaining issues, I noticed that this has been slipped through our attention. 
ROS uses a metric system and for coordinates, it uses [meter].
If the position of the AR marker is 2.0 meter away from manipulator, it is too far and the Inverse Kinematics(IK) of manipulator to reach to the marker cannot be generated/solved.

EndlessLoops, 2021-07-01:
Thanks for your rely.
After I tested it many times, this error occasionally appeared, even if the position of ar was within the grasping range of the arm.Sometimes it can't move to grap the ar pose in the demo testing and it show the error.I think the reason is that the feedback coordinates of ar is sometimes inaccurate.

closed 2021-07-01 by EndlessLoops

ROBOTIS-Will, 2021-07-01:
@EndlessLoops 
Thank you for the updates.
Due to the limited degree of freedom (4DOF + Gripper), certain pose that needs yaw rotation of the gripper cannot be made even within the range of the arm.
In order to solve IK for most cases, at least 6 DOF is required.
[image]
Objects in the video are aligned so that the gripper can solve the IK to reach the object.