ROBOTIS-GIT/open_manipulator #180: (Turtlebot & Open_manipulator) Error when launching open_manipulator_x_controller

DoctorJgriff, 2020-07-28:
I'm working in **ROS2 Dashing Diademata** with the TurtleBot3 **WafflePi**, it has a **RaspberryPi 3** with **ubuntu-18.04.3-preinstalled-server-arm64+raspi3** installed. I'm also using an **OpenCR board**.
My remote PC is using **Linux Mint 19** 

These are the instructions I was following:
https://emanual.robotis.com/docs/en/platform/openmanipulator_x/ros2_controller_package/#ros-controller-package

In section 13.1 of those instructions, I am told to run the command:
ros2 launch open_manipulator_x_controller open_manipulator_x_controller.launch.py usb_port:=/dev/ttyACM0

_The following is what is printed_
**ubuntu@ubuntu:~/robotis_ws$** ros2 launch open_manipulator_x_controller open_manipulator_x_controller.launch.py usb_port:=/dev/ttyACM0
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2020-07-28-17-49-53-272212-ubuntu-8216
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [open_manipulator_x_controller-1]: process started with pid [8227]
[ERROR] [open_manipulator_x_controller-1]: process has died [pid 8227, exit code -11, cmd '/home/ubuntu/robotis_ws/install/open_manipulator_x_controller/lib/open_manipulator_x_controller/open_manipulator_x_controller /dev/ttyACM0 1000000 __node:=open_manipulator_x_controller __params:=/home/ubuntu/robotis_ws/install/open_manipulator_x_controller/share/open_manipulator_x_controller/param/open_manipulator_x_controller_params.yaml'].

Upon running the actual command which failed:
**ubuntu@ubuntu:~/robotis_ws$** /home/ubuntu/robotis_ws/install/open_manipulator_x_controller/lib/open_manipulator_x_controller/open_manipulator_x_controller /dev/ttyACM0 1000000 __node:=open_manipulator_x_controller __params:=/home/ubuntu/robotis_ws/install/open_manipulator_x_controller/share/open_manipulator_x_controller/param/open_manipulator_x_controller_params.yaml

**I receive the following error output**:
port_name and baud_rate are set to /dev/ttyACM0, 1000000 
[ERROR] [TxRxResult] There is no status packet!
[ERROR] Please check your Dynamixel ID
[ERROR] [TxRxResult] There is no status packet!
[ERROR] Please check your Dynamixel ID
[ERROR] [TxRxResult] There is no status packet!
[ERROR] Please check your Dynamixel ID
[ERROR] [TxRxResult] There is no status packet!
[ERROR] Please check your Dynamixel ID
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [TxRxResult] There is no status packet!
[ERROR] Please check your Dynamixel ID
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelWorkbench] Failed to change torque status!
[ERROR] [DynamixelWorkbench] Failed to change torque status!
[ERROR] [DynamixelWorkbench] Failed to change torque status!
[ERROR] [DynamixelWorkbench] Failed to change torque status!
[ERROR] [DynamixelWorkbench] Failed to change torque status!
[ERROR] [TxRxResult] There is no status packet!
[ERROR] groupSyncRead getdata failed
[ERROR] groupSyncRead getdata failed
[ERROR] groupSyncRead getdata failed
Segmentation fault (core dumped)

ROBOTIS-Will, 2020-08-04:
Hi,

The eManual you are referring is written for a separate OpenMANIPULATOR-X product.
If you are using OpenMANIPULATOR-X on TurtleBot3 platform, please refer to the [Manipulation section](https://emanual.robotis.com/docs/en/platform/turtlebot3/manipulation/#manipulation) in ROS1.
We are currently working on some missing packages in ROS2 Manipulation.
Thank you.

ROBOTIS-Will, 2021-03-05:
Closing this issue as there isn't any recent activity. Please feel free to reopen when necessary.

closed 2021-03-05 by ROBOTIS-Will

cardboardcode, 2022-03-14:
> Hi,
>
> The eManual you are referring is written for a separate OpenMANIPULATOR-X product. If you are using OpenMANIPULATOR-X on TurtleBot3 platform, please refer to the [Manipulation [...]

Hi @ROBOTIS-Will, thank you for providing this amazing package.

**Enquiry**: Has this been addressed yet in **ROS2 Foxy**?

Encountered the same issue in setting up `open_manipulator` as @DoctorJgriff .

# Issue Description
**Encountered the following error** after setting up the necessary ROS2 packages and attempting to run the following command:

```bash
ros2 launch open_manipulator_x_controller open_manipulator_x_controller.launch.py usb_port:=/dev/ttyACM0
```

# Steps to Reproduce Error
```bash
cd $HOME
mkdir -p colcon_ws/src
cd colcon_ws/src
# Install ROS2 Foxy packages following instructions given in the following link:
# https://emanual.robotis.com/docs/en/platform/openmanipulator_x/quick_start_guide/#install-ros-on-pc
cd ~/colcon_ws
colcon build --symlink-install
source install/setup.bash
ros2 launch open_manipulator_x_controller open_manipulator_x_controller.launch.py usb_port:=/dev/ttyACM0
```
The error below is reported:
```bash
~/colcon_ws$ ros2 launch open_manipulator_x_controller open_manipulator_x_controller.launch.py
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2022-03-14-14-49-26-428301-turtlebot3-2180
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [open_manipulator_x_controller-1]: process started with pid [2182]
[ERROR] [open_manipulator_x_controller-1]: process has died [pid 2182, exit code -11, cmd '/home/ubuntu/colcon_ws/install/open_manipulator_x_controller/lib/open_manipulator_x_controller/open_manipulator_x_controller /dev/ttyUSB0 1000000 --ros-args -r __node:=open_manipulator_x_controller --params-file /home/ubuntu/colcon_ws/install/open_manipulator_x_controller/share/open_manipulator_x_controller/param/open_manipulator_x_controller_params.yaml'].
```
Ran the same commands as @DoctorJgriff did:
```bash
/home/ubuntu/colcon_ws/install/open_manipulator_x_controller/lib/open_manipulator_x_controller/open_manipulator_x_controller /dev/ttyACM0 1000000
```
The error below is reported:
```bash
[ERROR] [TxRxResult] There is no status packet!
[ERROR] Please check your Dynamixel ID
[ERROR] [TxRxResult] There is no status packet!
[ERROR] Please check your Dynamixel ID
[ERROR] [TxRxResult] There is no status packet!
[ERROR] Please check your Dynamixel ID
[ERROR] [TxRxResult] There is no status packet!
[ERROR] Please check your Dynamixel ID
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [TxRxResult] There is no status packet!
[ERROR] Please check your Dynamixel ID
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelWorkbench] Failed to change torque status!
[ERROR] [DynamixelWorkbench] Failed to change torque status!
[ERROR] [DynamixelWorkbench] Failed to change torque status!
[ERROR] [DynamixelWorkbench] Failed to change torque status!
[ERROR] [DynamixelWorkbench] Failed to change torque status!
[ERROR] [TxRxResult] There is no status packet!
[ERROR] groupSyncRead getdata failed
[ERROR] groupSyncRead getdata failed
[ERROR] groupSyncRead getdata failed
Segmentation fault (core dumped)

```

Yet to verify if running ROS1 will allow a workaround for this ROS2 issue. Will update here once I have done so.

ROBOTIS-Will, 2022-03-16:
hi @cardboardcode 
In case of running the OpenMANIPULATOR-X with your PC, make sure that you have correctly selected the USB-serial interface.
Depending on the interface, you will need to use different command as below.

- When operating with U2D2
```
$ ros2 launch open_manipulator_x_controller open_manipulator_x_controller.launch.py
```
- When operating with OpenCR
```
$ ros2 launch open_manipulator_x_controller open_manipulator_x_controller.launch.py usb_port:=/dev/ttyACM0
```

Aki1608, 2023-06-21:
Hi @ROBOTIS-Will, this behavior is very random for me. I am getting this error very randomly and frequently. Sometimes this keeps happening until I restart the board couple of time and sometimes it works on first try. I am using U2D2 board and the command you mentioned for U2D2 board. Is there any particular reason why this is happening? I cant able to pin point why it is happening so randomly.