ROBOTIS-GIT/open_manipulator #181: OpenManipulator-x controlled by PLc

LuchianRazvanAdrian, 2020-08-04:
Hello,
I want to control the openmanipulator-x using a PLC from Siemens and the OpenCR. What do I mean. For example, the PLC will send a digital command to OpenCR and let's say gripper will open up. How can I do that? Any advise?

ROBOTIS-Will, 2020-08-10:
Hi,
You can either use GPIO pins on OpenCR to receive the digital signal or use the UART or CAN port to receive the digital packet.
After receiving and parsing the packet from the PLC, you can program the OpenCR to move the OpenMANIPULATOR-X to a predetermined end effector pose based on Task Space.

LuchianRazvanAdrian, 2020-08-12:
Do you have a simple example? Let's say when I receive digital input on GPIO, the robots has to move to x, y, z coordinates. Can you help me with that?

LuchianRazvanAdrian, 2020-08-13:
`#include <open_manipulator_libs.h>
#include "Arduino.h"

OpenManipulator open_manipulator;
bool platform_state = true;
double control_time = 0.010;  //default control frequency (100Hz)
double present_time = 0.0;
double previous_time = 0.0;

void setup() {
Serial.begin(57600);
  while (!Serial);  // Wait until Serial is Opened

  open_manipulator.setOpenManipulatorCustomJointId(11, 12, 13, 14, 15); // ID 11, 12, 13, 14, 15 is default
  open_manipulator.initOpenManipulator(platform_state);
  Serial.println("OpenManipulator Init Begin");

  initDemo();
  
  Serial.println("Press [SW1] to execute move 1. Press [SW2] to execute move 2.");
}

void loop() {

    present_time = millis()/1000.0;

    // Trajectory following movement occurs here
    if(present_time-previous_time >= control_time)
    {
      open_manipulator.processOpenManipulator(millis()/1000.0);
      previous_time = millis()/1000.0;
    }

    // Read the next Pose to move
    if(digitalRead(BDPIN_PUSH_SW_1))
    {
      warning();
      move1(&open_manipulator);
    }
    else if(digitalRead(BDPIN_PUSH_SW_2))
    {
      warning();
      move2(&open_manipulator); 
    }
}

void move1(OpenManipulator *open_manipulator)
{
  moveJS(open_manipulator, 0.0, -1.0, 0.2, 0.8, 0.0, 10.0);
}

void move2(OpenManipulator *open_manipulator)
{
  moveJS(open_manipulator, -0.5, 0.0, 1.0, -1.0, 0.01, 10.0);
}

void initDemo()
{
  pinMode(BDPIN_PUSH_SW_1, INPUT);
  pinMode(BDPIN_PUSH_SW_2, INPUT);
  pinMode(BDPIN_GPIO_1, OUTPUT);
  digitalWrite(BDPIN_GPIO_1, LOW);
  Serial.println("OpenManipulator Init Completed");
}

void moveJS(OpenManipulator *open_manipulator, double j1, double j2, double j3, double j4, double gripper_pos, double time)
{
  static std::vector <double> target_angle;
  target_angle.clear();
  target_angle.push_back(j1);
  target_angle.push_back(j2);
  target_angle.push_back(j3);
  target_angle.push_back(j4);
  open_manipulator->makeJointTrajectory(target_angle, time);
  open_manipulator->makeToolTrajectory("gripper", gripper_pos);
}

void warning()
{
  Serial.println();
  Serial.println("WARNING!!! OpenMANIPULATOR-X operates in 5 seconds.");
  delay_ms(1000);+
  Serial.println("WARNING!!! OpenMANIPULATOR-X operates in 4 seconds.");
  delay_ms(1000);
  Serial.println("WARNING!!! OpenMANIPULATOR-X operates in 3 seconds.");
  delay_ms(1000);
  Serial.println("WARNING!!! OpenMANIPULATOR-X operates in 2 seconds.");
  delay_ms(1000);
  Serial.println("WARNING!!! OpenMANIPULATOR-X operates in 1 seconds.");
  open_manipulator.receiveAllJointActuatorValue();
  open_manipulator.receiveAllToolActuatorValue();
  open_manipulator.enableAllActuator();
  delay_ms(1000);
}`

I wrote this code, but I have a problem. When the robot starts moving, he moves too fast, even if the time for each move is set to 10 seconds.

ROBOTIS-Will, 2020-08-27:
Hi,
Your loop function does not run properly as it keeps executing the warning() and move1 or move2 function.
If you referred to the open_manipulator_chain_teaching example, please notice in the line 288 that there are moving state check and flag to prevent sending the trajectory too quickly.

ROBOTIS-Will, 2021-03-05:
Closing this issue as there isn't any recent activity. Please feel free to reopen when necessary.

closed 2021-03-05 by ROBOTIS-Will