ROBOTIS-GIT/open_manipulator #182: Can't control Gripper in chain_teaching on Arduino

Wolfman038, 2020-08-05:
Hi I'm having an issue where in the chain_teaching example on Arduino.  When I save a pose in the serial monitor, it registers that the gripper servo is moving, however when i run the movements, the gripper does not move at all. I know the gripper servo still works, because when the movements start, the motor returns to 0 position so idk why the servo wont move.

ROBOTIS-Will, 2020-08-10:
Hi,
There are min/max limits for the gripper setting. If the gripper value is greater than 0.010 or smaller than -0.010, gripper will not operate.
Also, please make sure that you have correct orientation for the gripper assembly especially aligning the horn marking.
[image]

ROBOTIS-Will, 2021-03-05:
Closing this issue as there isn't any recent activity. Please feel free to reopen when necessary.

closed 2021-03-05 by ROBOTIS-Will