ROBOTIS-GIT/open_manipulator #183: Cannot install ROS Packages

Wolfman038, 2020-08-07:
When I use the specific command in Ubuntu's Terminal: 
$ sudo apt-get install ros-kinetic-ros-controllers ros-kinetic-gazebo* ros-kinetic-moveit* ros-kinetic-industrial-core
I get a series of results saying:
Depends: ros-kinetic-roslib but it is not going to be installed
                             Depends: ros-kinetic-rospy but it is not going to be installed
                             Depends: ros-kinetic-rostopic but it is not going to be installed
 ros-kinetic-rqt-py-console : Depends: ros-kinetic-python-qt-binding (>= 0.2.19) but it is not going to be installed
                              Depends: ros-kinetic-qt-gui but it is not going to be installed
                              Depends: ros-kinetic-qt-gui-py-common but it is not going to be installed
                              Depends: ros-kinetic-rospy but it is not going to be installed
 ros-kinetic-rqt-py-trees : Depends: python-pygraphviz but it is not installable
                            Depends: python-termcolor but it is not installable
                            Depends: ros-kinetic-geometry-msgs but it is not going to be installed
                            Depends: ros-kinetic-py-trees but it is not going to be installed
                            Depends: ros-kinetic-py-trees-msgs but it is not going to be installed
                            Depends: ros-kinetic-qt-dotgraph but it is not going to be installed
                            Depends: ros-kinetic-rospy but it is not going to be installed
                            Depends: ros-kinetic-unique-id but it is not going to be installed
 ros-kinetic-rqt-reconfigure : Depends: ros-kinetic-dynamic-reconfigure but it is not going to be installed
                               Depends: ros-kinetic-python-qt-binding (>= 0.2.19) but it is not going to be installed
                               Depends: ros-kinetic-rospy but it is not going to be installed
 ros-kinetic-rqt-robot-dashboard : Depends: ros-kinetic-diagnostic-msgs but it is not going to be installed
                                   Depends: ros-kinetic-python-qt-binding (>= 0.2.19) but it is not going to be installed
                                   Depends: ros-kinetic-qt-gui but it is not going to be installed
                                   Depends: ros-kinetic-rospy but it is not going to be installed

Any tips?

ROBOTIS-Will, 2020-08-10:
Hi,
It looks like your ROS has not been installed properly.
Which version of Ubuntu and ROS did you install?
If you are running Ubuntu 16.04, you can use below command to install ROS on your PC.
```
$ sudo apt-get update && sudo apt-get upgrade
$ wget https://raw.githubusercontent.com/ROBOTIS-GIT/robotis_tools/master/install_ros_kinetic.sh && chmod 755 ./install_ros_kinetic.sh && bash ./install_ros_kinetic.sh
```

Wolfman038, 2020-08-10:
Hello, 
I used this command to install ROS again and then used the Install ROS Packages in e-manual 5.3 afterwards:
$ sudo apt-get install ros-kinetic-ros-controllers ros-kinetic-gazebo* ros-kinetic-moveit* ros-kinetic-industrial-core
to which i got the exact same error codes as before.

ROBOTIS-Will, 2020-08-12:
Hi,
ROS 1 Kinetic works best with Ubuntu 16.04.
Ubuntu 20.04 supports ROS 1 Noetic, but we are currently working on updating and releasing Noetic packages, and this may take several weeks.
I'd recommend to use Ubuntu 16.04 or 18.04 with ROS 1 Melodic.
Thank you.

Wolfman038, 2020-08-12:
understood.  I'll try that.

ROBOTIS-Will, 2021-03-05:
Closing this issue as there isn't any recent activity. Please feel free to reopen when necessary.

closed 2021-03-05 by ROBOTIS-Will