ROBOTIS-GIT/open_manipulator #188: Moveit2 support

pablogs9, 2020-10-02:
Hello, recently there have been a MoveIt release for ROS 2 Foxy: https://github.com/ros-planning/moveit2

I was wondering if you have plans for supporting openmanipulator in it. Also, if there is any tutorial or procedure for incorporating this support I would be happy to contribute.

Thanks!

ROBOTIS-Will, 2020-10-12:
Hi @pablogs9 
Thank you for your interest in OpenMANIPULATOR.
Yes, we will spend some time to support the latest ROS 1 and ROS 2 versions and new applications like MoveIt2.
And your contribution is highly welcomed and appreciated! :+1: 
For ROS 2 Foxy, you can use `ros2` branch as a starting point and when we receive PRs, we will review and merge them into the `foxy-devel` branch for preparing binary release in the future.
Thank you.

ROBOTIS-Will, 2021-03-05:
Closing this issue as there isn't any recent activity. Please feel free to reopen when necessary.

closed 2021-03-05 by ROBOTIS-Will