ROBOTIS-GIT/open_manipulator #189: control robot using troque mode

haocou, 2020-11-03:
Hello,
I want to control the openManipulator-X using torque(current) mode. The style of Dynamixel X-Series servo motors is XM430-W350.

But, the default mode is the position control mode. The Would you mind giving me some tips to control the robot in current(torque) mode? Thank you.

The execution process:
roslaunch open_manipulator_controller open_manipulator_controller.launch
roslaunch open_manipulator_control_gui open_manipulator_control_gui.launch

ROBOTIS-Will, 2020-11-04:
Hi @haocou ,
You might want to take a look at the [gravity compensation example](https://emanual.robotis.com/docs/en/platform/openmanipulator_x/ros_controls/#gravity-compensation).
This example uses effort joint interface which could be of your interest.
Thank you.

ROBOTIS-Will, 2021-03-05:
Closing this issue as there isn't any recent activity. Please feel free to reopen when necessary.

closed 2021-03-05 by ROBOTIS-Will