ROBOTIS-GIT/open_manipulator #190: [ERROR] [sr]fail to solve inverse kinematics (please change the solver) for 7-DoF open-manipulator-p

jangerritha, 2020-11-21:
Hello everyone,

I am working on an open-manipulator-p setup that has a third party end-effector mounted, which has an additional DoF. Following from this our kinematic chain contains 7-DoF. Three of the IK-solvers in **/open_manipulator_p_libs/src/kinematics.cpp** behave really strange for this setup and can't find solutions. One of the solvers Forward: ChainRule, Inverse: SRJacobianOnlyPosition works good out of the box. As I am pretty new to robotics, I am searching for paper references, but I was not able to determine any. Can anyone give a paper reference on which the implementations are based?

jangerritha, 2020-11-23:
@towardthesea tagged

ROBOTIS-Will, 2021-03-05:
Closing this issue as there isn't any recent activity. Please feel free to reopen when necessary.

closed 2021-03-05 by ROBOTIS-Will

towardthesea, 2021-03-08:
Hi @ROBOTIS-Will ,
We are still waiting for a response from the Open-Manipulator-P developers, so please keep it open and kindly tag who from Robotis taking charge of this issue.

Thanks!

ROBOTIS-Will, 2021-03-08:
Hi @towardthesea 
I'm afraid that we do not provide a specific guideline for a customized product.
Please note that the kinematics solution is designed based on 6DOF OpenMANIPULATOR-P.
If you are running on ROS, using the [MoveIt](https://moveit.ros.org/) tool might help you to operate your solution.
Thank you.