ROBOTIS-GIT/open_manipulator #193: Moveit : Failed to rotate in yaw direction!

ZhengQiushi, 2020-12-13:
hi, I met problem when running moveit with open manipulator in rviz. It seems that the arm can not rotate in the yaw direction, which means it can't rotate from left to right? I wonder how can I solve this problem?

ROBOTIS-Will, 2021-03-05:
Hi @ZhengQiushi 
Sorry about the delayed response.
MoveIt in RViz is designed for 6 DOF manipulator, and solving 4DOF OpenMANIPULATOR-X can be limited by MoveIt.
For more flexibility, you may want to use the [GUI program](https://emanual.robotis.com/docs/en/platform/openmanipulator_x/ros_operation/#gui-program).

yun-goon, 2024-12-19:
This issue will be closed due to prolonged inactivity. Please try again with the Humble or Noetic versions, and if the issue persists, feel free to reopen it.

closed 2024-12-19 by yun-goon