ROBOTIS-GIT/open_manipulator #197: 2 open manipulator x, same computer

LuchianRazvanAdrian, 2021-01-19:
Hello,
I have 2 open manipulator x robots controlled by 2 opencr 1.0. I want to use the same pc with ros and to control both robots with this pc. I managed to run the open manipulator controller script for both robots, on ACM0 and ACM1, and enable the robots. Also, I have a pick and place script. I doubled my script, but if i run both script, only one robots is moving. How can I control them using the same computer? What changes do I have to do?

ROBOTIS-Will, 2021-03-05:
Hi @LuchianRazvanAdrian,
I'm sorry about the delayed response.
You'll need to separate the controller for each manipulator.
Duplicate the controller launch file and use different namespaces for each controller will work (I tested with Gazebo quite a while ago)
[image]

yun-goon, 2024-12-19:
This issue appears to be resolved, so it will be closed.

closed 2024-12-19 by yun-goon