ROBOTIS-GIT/open_manipulator #199: Problem stablihing communication between U2D2 and ROS2 OpenMANIPULATOR

lkombo1, 2021-02-10:
Please find description error below:Section 1.4.1.2 USB latency Timer setting
Step 1) 
root@lucio-VirtualBox:~/robotis_ws/src# ros2 run open_manipulator_x_controller create_udev_rules

This script copies the udev rule to /etc/udev/rules.d/
to configure the U2D2 device for the OpenMANIPULATOR.
Reload rules

Section 1.1 Launch Controller
Step 2)
root@lucio-VirtualBox:~/robotis_ws/src# . /opt/ros/dashing/setup.bash
root@lucio-VirtualBox:~/robotis_ws/src# ros2 launch open_manipulator_x_controller open_manipulator_x_controller.launch.py
[INFO] [launch]: All log files can be found below /root/.ros/log/2021-02-08-15-38-18-469307-lucio-VirtualBox-27261
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [open_manipulator_x_controller-1]: process started with pid [27271]
[ERROR] [open_manipulator_x_controller-1]: process has died [pid 27271, exit code -11, cmd '/root/robotis_ws/src/install/open_manipulator_x_controller/lib/open_manipulator_x_controller/open_manipulator_x_controller /dev/ttyUSB0 1000000 __node:=open_manipulator_x_controller __params:=/root/robotis_ws/src/install/open_manipulator_x_controller/share/open_manipulator_x_controller/param/open_manipulator_x_controller_params.yaml'].

ROBOTIS-Will, 2021-02-17:
Hi @lkombo1 
It is not recommended to run ROS on virtual machine as it is difficult to identify the problem.
Could you describe more details about your system environment?

ROBOTIS-Will, 2021-07-01:
Closing the issue as there isn't recent activity. Please feel free to reopen when necessary.

closed 2021-07-01 by ROBOTIS-Will