ROBOTIS-GIT/open_manipulator #200: [ERROR] [checkJointLimit] Goal value exceeded limit at joint2.

suyunas, 2021-04-04:
roslaunch open_manipulator_controller open_manipulator_controller.launch usb_port:=/dev/ttyACM0 baud_rate:=1000000

The following error is caused when using the GUI with the above command whilst using this repository:

port_name and baud_rate are set to /dev/ttyACM0, 1000000 
Joint Dynamixel ID : 11, Model Name : XM430-W350
Joint Dynamixel ID : 12, Model Name : XM430-W350
Joint Dynamixel ID : 13, Model Name : XM430-W350
Joint Dynamixel ID : 14, Model Name : XM430-W350
Gripper Dynamixel ID : 15, Model Name :XM430-W350
[INFO] Succeeded to init /open_manipulator_controller
[ERROR] [checkJointLimit] Goal value exceeded limit at joint2.
[ERROR] [checkJointLimit] Goal value exceeded limit at joint2.
[ERROR] [checkJointLimit] Goal value exceeded limit at joint2.
[ERROR] [OpenManipulator Chain Custom]fail to solve inverse kinematics
[ERROR] [TASK_TRAJECTORY] Fail to solve IK
[ERROR] [OpenManipulator Chain Custom]fail to solve inverse kinematics
[ERROR] [TASK_TRAJECTORY] Fail to solve IK

I am using Ubuntu 16.04 with ROS Kinetic on openmanipulator comnnected to opencr using USB connection. if anybody can help??

ROBOTIS-Will, 2021-04-05:
Hi @suyunas 
Looks like the Joint2 is complaining about exceeding the operating range.
https://github.com/ROBOTIS-GIT/open_manipulator/blob/master/open_manipulator_libs/src/open_manipulator.cpp#L64

Please make sure that each joint is assembled correct.
When running RViz, you should get the identical pose of the physical manipulator and the 3D visualization on RViz.

Please set the manipulator as below before launching the controller to avoid any physical collision or error.
[image]

suyunas, 2021-04-05:
great. by placing the open manipulator in the initial position. i got everything right. Thank you

closed 2021-04-07 by suyunas