ROBOTIS-GIT/open_manipulator #201: sudo apt-get install ros-kinetic-compressed-image-transport ros-kinetic-camera-info-manager ros-kinetic-ar-track-alvar ros-kinetic-ar-track-alvar-msgs ros-kinetic-image-proc

suyunas, 2021-04-05:
sudo apt-get install ros-kinetic-compressed-image-transport ros-kinetic-camera-info-manager ros-kinetic-ar-track-alvar ros-kinetic-ar-track-alvar-msgs ros-kinetic-image-proc
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ros-kinetic-compressed-image-transport
E: Unable to locate package ros-kinetic-camera-info-manager
E: Unable to locate package ros-kinetic-ar-track-alvar
E: Unable to locate package ros-kinetic-ar-track-alvar-msgs
E: Unable to locate package ros-kinetic-image-proc

the error occured while runnign 5. (2nd line) in 9.1.1.3 Raspberry Pi Camera V2 on the following link:
https://emanual.robotis.com/docs/en/platform/openmanipulator_x/ros_perceptions/#astra-pro

ROBOTIS-Will, 2021-04-06:
Hi @suyunas 
I'm sorry about the confusion.
Some of these packages are already installed in [TurtleBot3 Kinetic version image for Raspberry Pi3B+](https://emanual.robotis.com/docs/en/platform/turtlebot3/sbc_setup/#sbc-setup) and don't have to be installed
They are, 
- ros-kinetic-compressed-image-transport
- ros-kinetic-camera-info-manager
and
- $ git clone https://github.com/UbiquityRobotics/raspicam_node.git

When installing ROS package to raspbian OS, you should use below method instead of using `apt install` command
http://wiki.ros.org/ROSberryPi/Installing%20ROS%20Kinetic%20on%20the%20Raspberry%20Pi#Adding_Released_Packages

We'll review this instruction document.
Thank you.

suyunas, 2021-04-07:
problem solved

closed 2021-04-07 by suyunas