ROBOTIS-GIT/open_manipulator #202: [ERROR] [OpenManipulator Chain Custom]fail to solve inverse kinematics [ERROR] [CUSTOM_TASK_TRAJECTORY] fail to solve IK

suyunas, 2021-04-07:
According to the tutorial while connecting Openmanipulator with OpenCR as follows:
https://emanual.robotis.com/docs/en/platform/openmanipulator_x/quick_start_guide/
I am able to use the teleop and gui appliaction. However on the gui, the manipulator is displaying the following errors while running the 'drawing' tab

[ERROR] [OpenManipulator Chain Custom]fail to solve inverse kinematics
[ERROR] [TASK_TRAJECTORY] Fail to solve IK

is there any thing missing on installation or need to change any file? many thanks
[image]

ROBOTIS-Will, 2021-04-08:
hi @suyunas 
The given pose of X, Y, Z for drawing line is exceeding the operating range and cannot be achieved.
Please try with smaller values first.

suyunas, 2021-04-08:
solved

closed 2021-04-08 by suyunas