ROBOTIS-GIT/open_manipulator #204: Connection to openmanipulator-X through raspberry pi

suyunas, 2021-04-08:
https://emanual.robotis.com/docs/en/platform/openmanipulator_x/ros_controller_package/

[image]
after running the 2nd command on PC i am having the following errors. Note that when i am using manipulator connected to the PC using a USB cable then it works perfectly but when attached it through raspberry pi 3 on turtlebot3, it gives the error stated below:
[image]

How can i connect to openmanipulator-X through raspberry pi through raspberry pi and not directly through opencr using usb cable? can somebody list the set of commands for pc and turtlebot3?

ROBOTIS-Will, 2021-04-09:
Hi, @suyunas 
If you have connected OpenMANIPULATOR-X directly to Raspberry Pi via U2D2, you should change the device port.
Since /dev/ttyACM0 is assigned to OpenCR and /dev/ttyUSB0 is assigned to LDS 1.5, most likely, your U2D2 should have assigned to `/dev/ttyUSB1`.
Please modify the OpenMANIPULATOR code to connect via assigned port.

suyunas, 2021-04-09:
i am not using U2D2. It seems that U2D2 is required to run the manipulator to establish a remote connection. Is there anyways i can control manipulator using the following strategy:

Remote PC--->RPi----->OpenCR---->Manipulator X???

ROBOTIS-Will, 2021-04-12:
Have you read this TurtleBot3 Manipulation eManual before?
https://emanual.robotis.com/docs/en/platform/turtlebot3/manipulation/#manipulation

suyunas, 2021-04-24:
yes i checked the manual. Basically the gui function called for open-manipulator +tb3 has missing drawing circle rhombas and line etc as shown in figure below:
/home/trustai/Pictures/12

I need to make use of those function.

suyunas, 2021-04-24:
yes i checked the emanual. Apparently the drawing and option tabs are available on the gui called for openmanipulator-X connected through openCR only as shown in figure below:

[image]

These tabs are not available on the gui called for openmanipulator-X connected through turtlebot3,
[image]
 i am using the later setup.

ROBOTIS-Will, 2021-04-26:
Both GUI software works in the same way.
You just need to send a proper topic so from your PC to control the OpenMANIPULATOR.
Since all hardware in ROS1 are connected in the same network, you don't necessarily have to deliver the topic from PC to RPi to OpenCR.

suyunas, 2021-04-27:
Basically i was trying to explore different options for the project. I will look into other ways for calling those functionalities.

closed 2021-04-27 by suyunas