ROBOTIS-GIT/open_manipulator #205: Unable to control open_manipulator using ROS2

shonigmann, 2021-04-19:
I am trying to control my open manipulator x through ROS2 Foxy, setting things up based on the instructions [here](https://emanual.robotis.com/docs/en/platform/openmanipulator_x/ros2_setup/#ros-setup). I can run `open_manipulator_x_controller.launch.py` without any errors, the motors engage and hold the current position, rviz runs fine and displays the current position and matches changes if I push the end effector around a bit. 

My issue is that I can't successfully **send** position commands. I can run `open_manipulator_x_teleop_keyboard` without error from both the remote PC and the RasPi, but pressing any of the keys doesn't do anything. I have also tried running some of the services (e.g.  `/goal_joint_space_path`) without any luck. 

**Setup details:**
Remote PC: Ubuntu 20.04 on Virtual Box with ROS2 Foxy
SBC: Raspi 3B+ running Ubuntu Server 20.04 with ROS2 Foxy
Hardware: Open Manipulator X -> OpenCR, flashed with usb_to_dxl.ino -> Raspi

Any help on this would be appreciated!

ROBOTIS-Will, 2021-04-21:
Hi @shonigmann 
Thank you for reporting the issue.
Current example is designed to run on Dashing, but we have a plan to update packages for the latest ROS support.
I'll review and update the teleop package.
Thanks!

sardanian, 2021-06-11:
I was interested in using this with foxy and am following this thread. Is this still planned for Q2? Thank you

ROBOTIS-Will, 2021-06-11:
Hi @sardanian 
Yes, this package is in progress of update to support Foxy and Galactic this month.
Thank you!

ROBOTIS-Will, 2021-07-01:
Hi @sardanian @shonigmann 
I'm sorry about the delay regarding ROS2 update.
ROS2 Foxy/Galactic support update is put on hold for a while due to the urgent update of some ROS1 Noetic packages, but I'll get back to ROS2 as soon as possible and hoping to be done by mid July.
Thank you for your patience.

sardanian, 2021-07-02:
Ok. Thank you. I’m looking forward to the update. I’m also happy to help if
I can be of help.

ROBOTIS-Will, 2021-07-05:
@sardanian 
At moment, MoveIt2 doesn't have ported setup assistant yet, so I might end up using the ROBOTIS controller and GUI instead of MoveIt2 and RViz2 with an interactive marker until the setup assistant is ready for ROS2.

sardanian, 2021-08-06:
Is there an update on timing for ROS2 foxy or galactic? Not trying to rush, but just looking for an idea on timing if possible. 

Thank you.

ROBOTIS-Will, 2021-08-09:
Hi @sardanian 
Thanks for the friendly ping :)
I've been working on the eManual documentation and will be released by the end of this week or sooner.

sardanian, 2021-09-07:
Hi there. I am checking in again on an update for the ros2 packages to updated versions. I noticed the Emanuel was updated to include noetic but I haven’t seen it for foxy/galactic.

Thank you.

ROBOTIS-Will, 2021-09-22:
Hi @sardanian 
I'm sorry about the delayed response.
I'm currently working with the Foxy update.
Once the basic operation with either the keyboard or GUI is completed, the eManual will be updated.
Thank you.

sardanian, 2021-10-20:
I see that the foxy version of this has been released. Is that accurate? Additionally, is there plans to go to Galaxy or should we jump into foxy at this point? Thank you.

closed 2024-12-19 by yun-goon