ROBOTIS-GIT/open_manipulator #206: TurtleBot3 Home Service Challenge

suyunas, 2021-04-24:
Whilst doing the TurtleBot3 Home Service Challenge. I am having the following error:
[image]

Somebody can help?

ROBOTIS-Will, 2021-04-26:
Have you run belows from your Remote PC?
```bash
$ chmod +x ~/tb3_ssh_keygen
$ ~/tb3_ssh_keygen 192.168.1.10
```
and the below from your Raspberry Pi on TurtleBot3?
```bash
$ chmod +x ~/env.bash
```
Did you get any warning or error messages when running the above?

suyunas, 2021-04-26:
[image]
[Remote PC]
Upper left: 
$ roscore

[Pi]
Upper right:
$ roslaunch turtlebot3_bringup turtlebot3_robot.launch

[Remote PC]
Middle left:
$ export TURTLEBOT3_MODEL=${TB3_MODEL}
$ roslaunch turtlebot3_manipulation_description turtlebot3_manipulation_view.launch use_gui:=true

[Remote PC]
Middle right:
$ nano ~/tb3_ssh_keygen
$ chmod +x ~/tb3_ssh_keygen
$ ~/tb3_ssh_keygen 192.168.43.191

[Pi]
Lower left:
$ nano ~/env.bash
$ chmod +x ~/env.bash

[Remote PC]
Lower right:
$ roslaunch turtlebot3_home_service_challenge_tools turtlebot3_pi_cam_remote.launch address:=192.168.43.220

ROBOTIS-Will, 2021-04-27:
when running the `tb3_ssh_keygen` command, you should use the IP address of TurtleBot3 SBC as below.

[Remote PC]
Middle right:
```bash
$ nano ~/tb3_ssh_keygen
$ chmod +x ~/tb3_ssh_keygen
$ ~/tb3_ssh_keygen 192.168.43.220
```

suyunas, 2021-04-27:
Apologies in typing i put  ~/tb3_ssh_keygen 192.168.43.191. However from the picture above it is clear that i have used ~/tb3_ssh_keygen 192.168.43.220. Kindly check

ROBOTIS-Will, 2021-04-29:
@suyunas 
I've tested with my Kinetic setup, but couldn't reproduce the issue.
Have you set the ROS_MASTER_URI and ROS_HOSTNAME for both Remote PC and Raspberry Pi?
[](https://emanual.robotis.com/docs/en/platform/turtlebot3/quick-start/#network-configuration)

suyunas, 2021-04-29:
nano ~/.bashrc configurations for Remote PC:
alias eb='nano ~/.bashrc'
alias sb='source ~/.bashrc'
alias gs='git status'
alias gp='git pull'
alias cw='cd ~/catkin_ws'
alias cs='cd ~/catkin_ws/src'
alias cm='cd ~/catkin_ws && catkin_make'
source /opt/ros/kinetic/setup.bash
source ~/catkin_ws/devel/setup.bash
export ROS_MASTER_URI=http://192.168.43.191:11311
export ROS_HOSTNAME=192.168.43.191
export TURTLEBOT3_MODEL=waffle_pi

TB3:
alias gp='git pull'
alias cw='cd ~/catkin_ws'
alias cs='cd ~/catkin_ws/src'
alias cm='cd ~/catkin_ws && catkin_make'
source /opt/ros/kinetic/setup.bash
source ~/catkin_ws/devel/setup.bash
export ROS_MASTER_URI=http://192.168.43.191:11311
export ROS_HOSTNAME=192.168.43.220

ROBOTIS-Will, 2021-04-30:
@suyunas 
Your network configuration looks correct.
> 'Be sure Remote PC and TurtleBot3 are connected under the same IP'. This statement is a bit confusing on Home Service Challenge. how can we keep IP same for two separate devices?

Sorry about the confusion. This means to configure your IP address in the same IP range, such as 192.168.43.xxx

TB3 HSC is currently supported with Kinetic which runs on Ubuntu 16.04 or the provided Raspbian OS image from us.

Please try below command and let me know. This will copy the SSH key to TurtleBot3 SBC.
```bash
$ ssh-copy-id pi@192.168.43.220
```

suyunas, 2021-04-30:
can be seen in the middle right:
[image]

yun-goon, 2024-12-19:
@suyunas Apologies for the delayed response. We have assembled a TF team to address OpenManipulator-x issues and are actively working on resolutions. Please note that we do not support EOL (End-of-Life) ROS versions; our support currently extends to Noetic and Humble versions. We kindly ask that you try again with the latest version. If the same issue arises, please feel free to resubmit it, and we will address it promptly.

closed 2024-12-19 by yun-goon