ROBOTIS-GIT/open_manipulator #208: OpenManipulator + ROS2 dashing - nothing is being published

Karolk99, 2021-05-27:
Hi, I am using OpenManipulator X and for that purpose I created docker images one with ros-kinetic and second one with ros2-dashing. The first one works correct, but the second one (with ros2-dashing) works incorrect.
I did instruction from manual step by step.I connect to the OpenManipulator X using ```ros2 launch open_manipulator_x_controller open_manipulator_x_controller.launch.py```  and get correct result. But when I try to: ```ros2 topic echo /kinematics_pose``` there is nothing published and also teleopkeyboard is not working becouse of that. I hope that you can help me :). Bellow there is my code:
Dockerfile:
```Dockerfile
FROM ubuntu:18.04

SHELL [ "/bin/bash", "-c" ]
RUN apt-get update && \
         get -y install sudo
RUN sudo apt update -y && sudo apt upgrade -y

ENV TZ=Europe/Warsaw

RUN apt-get clean && apt-get update && apt-get install -y locales
RUN DEBIAN_FRONTEND="noninteractive" apt-get -y install tzdata

COPY install_ros2_dashing.sh .
RUN chmod 755 ./install_ros2_dashing.sh && bash ./install_ros2_dashing.sh

RUN source ~/.bashrc
RUN sudo apt install -y ros-dashing-python* ros-dashing-rqt* ros-dashing-joint-state-publisher
RUN cd ~/colcon_ws/src/  && \
    git clone -b ros2 https://github.com/ROBOTIS-GIT/DynamixelSDK.git && \ 
    git clone -b ros2 https://github.com/ROBOTIS-GIT/dynamixel-workbench.git && \  
    git clone -b ros2 https://github.com/ROBOTIS-GIT/open_manipulator.git  && \
    git clone -b ros2 https://github.com/ROBOTIS-GIT/open_manipulator_msgs.git && \ 
    git clone -b ros2 https://github.com/ROBOTIS-GIT/open_manipulator_dependencies.git && \ 
    git clone -b ros2 https://github.com/ROBOTIS-GIT/robotis_manipulator.git

COPY ros_catkin_entrypoint.sh /app/
RUN chmod +x /app/ros_catkin_entrypoint.sh
CMD ["/app/ros_catkin_entrypoint.sh"]
```
run script:
```bash
xhost local:root

docker run -it --privileged --rm \
--runtime=nvidia \
--env=NVIDIA_VISIBLE_DEVICES=all \
--env=NVIDIA_DRIVER_CAPABILITIES=all \
--env=QT_X11_NO_MITSHM=1 \
--env DISPLAY=$DISPLAY \
-v /tmp/.X11-unix:/tmp/.X11-unix \
--device /dev/ttyUSB0 \
--device /dev/net/tun \
ros2_arm /bin/bash
```
then in terminal i type:
```bash
cd ~/colcon_ws && colcon build --symlink-install
source ~/.bashrc

ros2 run open_manipulator_x_controller create_udev_rules

export RCUTILS_CONSOLE_STDOUT_LINE_BUFFERED=1
ros2 launch open_manipulator_x_controller open_manipulator_x_controller.launch.py &
```

***there some warnings when ```cd ~/colcon_ws && colcon build --symlink-install```
```logs
Summary: 13 packages finished [33.7s]
  3 packages had stderr output: dynamixel_workbench_toolbox open_manipulator_x_controller open_manipulator_x_libs
```

Karolk99, 2021-05-27:
all logs from ```cd ~/colcon_ws && colcon build --symlink-install```
```logs
root@3e363eb3c620:/# cd ~/colcon_ws && colcon build --symlink-install
Starting >>> dynamixel_sdk
Starting >>> cmake_modules
Starting >>> open_manipulator_msgs
Starting >>> dynamixel_sdk_custom_interfaces
Starting >>> open_manipulator_x_description
Finished <<< cmake_modules [3.12s]                                                                                                   
Starting >>> robotis_manipulator
Finished <<< open_manipulator_x_description [3.12s]
Finished <<< dynamixel_sdk [3.91s]                                                                             
Starting >>> dynamixel_workbench_toolbox                                                                                     
--- stderr: dynamixel_workbench_toolbox                                                                                              
/root/colcon_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/src/dynamixel_workbench_toolbox/dynamixel_driver.cpp: In member function 'bool DynamixelDriver::readRegister(uint8_t, uint16_t, uint16_t, uint32_t*, const char**)':
/root/colcon_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/src/dynamixel_workbench_toolbox/dynamixel_driver.cpp:776:27: warning: ISO C++ forbids variable length array 'data_read' [-Wvla]
   uint8_t data_read[length];
                           ^
/root/colcon_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/src/dynamixel_workbench_toolbox/dynamixel_driver.cpp: In member function 'bool DynamixelDriver::syncWrite(uint8_t, uint8_t*, uint8_t, int32_t*, uint8_t, const char**)':
/root/colcon_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/src/dynamixel_workbench_toolbox/dynamixel_driver.cpp:1012:49: warning: ISO C++ forbids variable length array 'multi_parameter' [-Wvla]
   uint8_t multi_parameter[4*data_num_for_each_id];
                                                 ^
---
Finished <<< dynamixel_workbench_toolbox [8.65s]
Starting >>> dynamixel_workbench
Finished <<< dynamixel_sdk_custom_interfaces [15.3s]                                                                                    
Starting >>> dynamixel_sdk_examples
Finished <<< dynamixel_workbench [2.87s]                                                                                 
Finished <<< robotis_manipulator [14.4s]                                                                                    
Starting >>> open_manipulator_x_libs
Finished <<< dynamixel_sdk_examples [4.01s]                                                                                      
Finished <<< open_manipulator_msgs [20.1s]                                                                                 
Starting >>> open_manipulator_x_teleop
Finished <<< open_manipulator_x_teleop [5.37s]                                                                                 
--- stderr: open_manipulator_x_libs                                   
/root/colcon_ws/src/open_manipulator/open_manipulator_x_libs/src/dynamixel.cpp: In member function 'bool dynamixel::JointDynamixel::write_goal_position(std::vector<unsigned char>, std::vector<double>)':
/root/colcon_ws/src/open_manipulator/open_manipulator_x_libs/src/dynamixel.cpp:270:38: warning: ISO C++ forbids variable length array 'id_array' [-Wvla]
   uint8_t id_array[actuator_id.size()];
                                      ^
/root/colcon_ws/src/open_manipulator/open_manipulator_x_libs/src/dynamixel.cpp:271:43: warning: ISO C++ forbids variable length array 'goal_position' [-Wvla]
   int32_t goal_position[actuator_id.size()];
                                           ^
/root/colcon_ws/src/open_manipulator/open_manipulator_x_libs/src/dynamixel.cpp: In member function 'std::vector<robotis_manipulator::_Point> dynamixel::JointDynamixel::receive_all_dynamixel_value(std::vector<unsigned char>)':
/root/colcon_ws/src/open_manipulator/open_manipulator_x_libs/src/dynamixel.cpp:295:38: warning: ISO C++ forbids variable length array 'id_array' [-Wvla]
   uint8_t id_array[actuator_id.size()];
                                      ^
/root/colcon_ws/src/open_manipulator/open_manipulator_x_libs/src/dynamixel.cpp:299:41: warning: ISO C++ forbids variable length array 'get_current' [-Wvla]
   int32_t get_current[actuator_id.size()];
                                         ^
/root/colcon_ws/src/open_manipulator/open_manipulator_x_libs/src/dynamixel.cpp:300:42: warning: ISO C++ forbids variable length array 'get_velocity' [-Wvla]
   int32_t get_velocity[actuator_id.size()];
                                          ^
/root/colcon_ws/src/open_manipulator/open_manipulator_x_libs/src/dynamixel.cpp:301:42: warning: ISO C++ forbids variable length array 'get_position' [-Wvla]
   int32_t get_position[actuator_id.size()];
                                          ^
/root/colcon_ws/src/open_manipulator/open_manipulator_x_libs/src/dynamixel.cpp: In member function 'bool dynamixel::JointDynamixelProfileControl::write_goal_profiling_control_value(std::vector<unsigned char>, std::vector<robotis_manipulator::_Point>)':
/root/colcon_ws/src/open_manipulator/open_manipulator_x_libs/src/dynamixel.cpp:606:38: warning: ISO C++ forbids variable length array 'id_array' [-Wvla]
   uint8_t id_array[actuator_id.size()];
                                      ^
[... cut ...]
/root/colcon_ws/src/open_manipulator/open_manipulator_x_controller/src/open_manipulator_x_controller.cpp:104:21: warning: unused variable 'name' [-Wunused-variable]
   for (auto const & name:tools_name)
                     ^~~~
---
Finished <<< open_manipulator_x_controller [6.56s]
Starting >>> open_manipulator_x
Finished <<< open_manipulator_x [1.12s]                     

Summary: 13 packages finished [33.7s]
  3 packages had stderr output: dynamixel_workbench_toolbox open_manipulator_x_controller open_manipulator_x_libs
```

ROBOTIS-Will, 2021-05-28:
Hi @Karolk99 
I'm not familiar with docker environment especially when running the actual platform.
Those compile warnings can be ignored as they are not critical in running the example.
CW from the top right corner, I've launched and keyboard operated my OpenMANIPULATOR-X.

1. ros2 launch open_manipulator_x_controller open_manipulator_x_controller.launch.py
2. ros2 launch open_manipulator_x_description open_manipulator_x_rviz.launch.py
3. rqt
4. ros2 run open_manipulator_x_teleop open_manipulator_x_teleop_keyboard

[image]

sardanian, 2021-06-11:
I built this in my ubuntu environment and am getting this same issue.

The keyboard operations are not working for me. I followed the steps that you provided and it is still not working for me. The arm does not move.

This is my build log. I am also geting the errors that seem to indicate some of the packages didn't compile correctly.

This is new to me. Any help is greatly appreciated.

Starting >>> dynamixel_sdk
Starting >>> cmake_modules
Starting >>> open_manipulator_msgs                                              
Starting >>> dynamixel_sdk_custom_interfaces
Starting >>> open_manipulator_x_description
Finished <<< cmake_modules [12.0s]                                                                       
Starting >>> robotis_manipulator
Finished <<< open_manipulator_x_description [12.2s]                                                            
Finished <<< dynamixel_sdk [15.6s]                                                                                
Starting >>> dynamixel_workbench_toolbox
--- stderr: dynamixel_workbench_toolbox                                          
/home/ronnie/colcon_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/src/dynamixel_workbench_toolbox/dynamixel_driver.cpp: In member function ‘bool DynamixelDriver::readRegister(uint8_t, uint16_t, uint16_t, uint32_t*, const char**)’:
/home/ronnie/colcon_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/src/dynamixel_workbench_toolbox/dynamixel_driver.cpp:776:27: warning: ISO C++ forbids variable length array ‘data_read’ [-Wvla]
   uint8_t data_read[length];
                           ^
/home/ronnie/colcon_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/src/dynamixel_workbench_toolbox/dynamixel_driver.cpp: In member function ‘bool DynamixelDriver::syncWrite(uint8_t, uint8_t*, uint8_t, int32_t*, uint8_t, const char**)’:
/home/ronnie/colcon_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/src/dynamixel_workbench_toolbox/dynamixel_driver.cpp:1012:49: warning: ISO C++ forbids variable length array ‘multi_parameter’ [-Wvla]
   uint8_t multi_parameter[4*data_num_for_each_id];
                                                 ^
---
Finished <<< dynamixel_workbench_toolbox [28.4s]
Starting >>> dynamixel_workbench
Finished <<< dynamixel_workbench [8.92s]                                         
Finished <<< dynamixel_sdk_custom_interfaces [54.1s]                             
Starting >>> dynamixel_sdk_examples
Finished <<< robotis_manipulator [58.3s]                                                   
Starting >>> open_manipulator_x_libs
Finished <<< dynamixel_sdk_examples [21.9s]                                                
[Processing: open_manipulator_msgs, open_manipulator_x_libs]                                                                          
                                                                                                                        Finished <<< open_manipulator_msgs [1min 50s]                                                              
Starting >>> open_manipulator_x_teleop
--- stderr: open_manipulator_x_libs                                                       
/home/ronnie/colcon_ws/src/open_manipulator/open_manipulator_x_libs/src/custom_trajectory.cpp: In member function ‘virtual void custom_trajectory::Line::setOption(const void*)’:
/home/ronnie/colcon_ws/src/open_manipulator/open_manipulator_x_libs/src/custom_trajectory.cpp:95:34: warning: unused parameter ‘arg’ [-Wunused-parameter]
 void Line::setOption(const void *arg) {}
                                  ^~~
/home/ronnie/colcon_ws/src/open_manipulator/open_manipulator_x_libs/src/custom_trajectory.cpp: In member function ‘virtual void custom_trajectory::Circle::setOption(const void*)’:
/home/ronnie/colcon_ws/src/open_manipulator/open_manipulator_x_libs/src/custom_trajectory.cpp:169:36: warning: unused parameter ‘arg’ [-Wunused-parameter]
 void Circle::setOption(const void *arg){}
                                    ^~~
/home/ronnie/colcon_ws/src/open_manipulator/open_manipulator_x_libs/src/custom_trajectory.cpp: In member function ‘virtual void custom_trajectory::Rhombus::setOption(const void*)’:
/home/ronnie/colcon_ws/src/open_manipulator/open_manipulator_x_libs/src/custom_trajectory.cpp:262:37: warning: unused parameter ‘arg’ [-Wunused-parameter]
 void Rhombus::setOption(const void *arg){}
                                     ^~~
/home/ronnie/colcon_ws/src/open_manipulator/open_manipulator_x_libs/src/custom_trajectory.cpp: In member function ‘virtual void custom_trajectory::Heart::setOption(const void*)’:
/home/ronnie/colcon_ws/src/open_manipulator/open_manipulator_x_libs/src/custom_trajectory.cpp:336:35: warning: unused parameter ‘arg’ [-Wunused-parameter]
 void Heart::setOption(const void *arg){}
                                   ^~~
/home/ronnie/colcon_ws/src/open_manipulator/open_manipulator_x_libs/src/kinematics.cpp: In member function ‘virtual void kinematics::SolverUsingCRAndJacobian::setOption(const void*)’:
/home/ronnie/colcon_ws/src/open_manipulator/open_manipulator_x_libs/src/kinematics.cpp:27:54: warning: unused parameter ‘arg’ [-Wunused-parameter]
 void SolverUsingCRAndJacobian::setOption(const void *arg){}
                                                      ^~~
/home/ronnie/colcon_ws/src/open_manipulator/open_manipulator_x_libs/src/kinematics.cpp: In member function ‘virtual void kinematics::SolverUsingCRAndSRJacobian::setOption(const void*)’:
[... cut ...]
/home/ronnie/colcon_ws/src/open_manipulator/open_manipulator_x_controller/src/open_manipulator_x_controller.cpp:104:21: warning: unused variable ‘name’ [-Wunused-variable]
   for (auto const & name:tools_name)
                     ^~~~
---
Finished <<< open_manipulator_x_controller [38.1s]
Starting >>> open_manipulator_x
Finished <<< open_manipulator_x [6.36s]                          

Summary: 13 packages finished [2min 49s]
  3 packages had stderr output: dynamixel_workbench_toolbox open_manipulator_x_controller open_manipulator_x_libs

sardanian, 2021-06-13:
So I tried this on my desktop and it works. However, when tried on a Jetson Xavier NX Ubuntu 18.04 it doesn't. Any thought on what the problem might be? Is there a reason this wouldn't work on a Jetson?

ROBOTIS-Will, 2021-06-14:
@sardanian 
Could you copy and paste the output from the Jetson?
What kind of error / symptom do you get from the system when running the package?

sardanian, 2021-06-15:
I was able to resolve it. Thank you.

yun-goon, 2024-12-19:
This issue appears to be resolved, so it will be closed.

closed 2024-12-19 by yun-goon