ROBOTIS-GIT/open_manipulator #209: Control OPEN Manipulator X with OpenCR or U2D2 + U2D2 PHB

slavenpetkovic, 2021-06-01:
I want to use one Turtlebot3 and Open Manipulator X separately. I have one OpenCR to control Turtlebot and I wan't to buy something to control OpenManipulator X with ROS. 
Is it possible to use DYNAMIXEL Starter Set to control OM-X or I have to buy another OPENCR?

ROBOTIS-Will, 2021-06-02:
Hi @slavenpetkovic 
If you are going to mount the OpenMANIPULATOR-X on top of TurtleBot3 Waffle Pi, you may want to refer to the eManual below.
https://emanual.robotis.com/docs/en/platform/turtlebot3/manipulation/#manipulation

If you are operating two robots separately, you can use DYNAMIXEL Starter Set to connect OpenMANIPULATOR-X to your PC.
Although OpenCR can be used as a [communication converter](https://emanual.robotis.com/docs/en/platform/openmanipulator_x/ros_setup/#communication-converter) for OpenMANIPULATOR-X, U2D2 + U2D2 PHB is recommended in terms of supporting higher baudrate, optimized for DYNAMIXEL Wizard 2.0 for DYNAMIXEL configuration, etc.
Thanks!

slavenpetkovic, 2021-06-02:
So i should use U2D2 + U2D2 PHB (DYNAMIXEL Starter Set)+ RPi to control Open Manipulator X and OpenCR to control turtlebot3 (without that OM X)?

ROBOTIS-Will, 2021-06-02:
@slavenpetkovic 
You said you are going to use TurtleBot3 and OM-X separately which I understood that you'll have a separate hardware and control for both system.
What's your plan for the entire system design?

slavenpetkovic, 2021-06-03:
Remote PC (roscore) + Tutrlebot3 with OpenCR and RPi + Open Manipulator X with U2D2 + U2D2 PHB + RPi.

slavenpetkovic, 2021-06-26:
Can you tell me, is this combination possible?

ROBOTIS-Will, 2021-06-29:
@slavenpetkovic 
Sorry about the delayed reply.
The current example only use OpenCR as a communication interface.
Controlling the OpenMANIPULATOR-X with RPi via U2D2 is possible, but may require some modification of the code and even creating a new package depending on your purpose.
You'll be most likely using a default TurtleBot3 Waffle Pi packages and OpenCR firmware along with [openmanipulator packages](https://emanual.robotis.com/docs/en/platform/openmanipulator_x/ros_setup/#install-ros-packages)

slavenpetkovic, 2021-06-29:
So the safer and better option to control OM-X with RPi is via OpenCR?

ROBOTIS-Will, 2021-06-29:
It depends.
If you are going to control TB3 and OM-X as separate modules that only shares SBC, U2D2 is doable with minor modifications.
If OM-X has to be linked to TB3 tf like TB3 manipulation example, OpenCR will be a better option as it already has an example.

slavenpetkovic, 2021-06-29:
To sum up, if I want to use Turtlebot3 alone (without OMX on top of it) to move and "deliver" something and in same system use OM-X fixed to ground to do some pick and place actions i can use this configuration : 

TB3 + OPENCR + RPi to control TB3 and OM-X with OPENCR + another RPi to control OM-X .. Both RPis are on the same network and all that is "controlled" by one ROS master PC.

ROBOTIS-Will, 2021-06-29:
@slavenpetkovic 

If TB3 is only delivering around in the map and OM-X loads items on TB3, you are correct, and there's a better way to do it as below.
OpenCR as a communication converter is an overkill unless you need to attach some other sensors or devices to it (which will require to modify the usb_to_dxl Arduino example code).

- TB3 + OpenCR + RPi (default TB3 package)
- OM-X + [DYNAMIXEL Starter Set](http://en.robotis.com/shop_en/item.php?it_id=902-0161-200) + RPi

Of course all RPi and master PC have to be connected in the same network so that ROS master can find all the nodes.

yun-goon, 2024-12-19:
This issue will be closed due to prolonged inactivity.

closed 2024-12-19 by yun-goon