ROBOTIS-GIT/open_manipulator #210: ROS2 simulation, perceptions and Controls

sardanian, 2021-06-13:
Hi there,

I was wondering if the simulation, perceptions (camera) and/or controls packages would be included with the new ROS2 versions coming out this month.

Thank you,
Ronnie

ROBOTIS-Will, 2021-06-14:
Hi @sardanian 
The very basic features to operate the manipulator will be updated first, then other examples or applications (such as perceptions) might be scheduled for the update based on users feedback and internal review.
Thank you.

yun-goon, 2024-12-19:
Hi, @sardanian 

Thank you for your inquiry.

This issue will be closed as new packages have been uploaded. The perception packages are scheduled to be updated soon.

Please try again with the Humble or Noetic versions, and if the issue persists, feel free to reopen it.

closed 2024-12-19 by yun-goon