ROBOTIS-GIT/open_manipulator #212: Turtlebot3 Home Service Challenge - remote roslaunch failed to launch:tb3

Prajwalmj, 2021-06-23:
* Hi, I am trying to use home service challenge to do pick and place task in a different map. I started following steps as in the [e-manual](https://emanual.robotis.com/docs/en/platform/turtlebot3/home_service_challenge/).
* I created ssh keys as mentioned in the e-manual, by providing correct IP address of turtlebot3, and both PC and turtlebot3 are connected under same IP address range.
* However, when I try to launch rpi camera using 
``` roslaunch turtlebot3_home_service_challenge_tools turtlebot3_pi_cam_remote.launch address:=192.168.0.153```
command, I get this error.
* I have also attached the image to show the commands executed and error.

[image]

* This issue is a bit similar to #206 issue, but error messages are different

Please let me know what is the issue, I appreciate the help!

ROBOTIS-Will, 2021-06-25:
hi @Prajwalmj 
You should create the ssh key with the IP address of your Remote PC where roscore is running.
```
$ ~/tb3_ssh_keygen 192.168.0.125
```

Prajwalmj, 2021-06-25:
* Hi @ROBOTIS-Will, in the manual, it said this:
 ```Execute the file with TurtleBot3’s IP address. Be sure Remote PC and TurtleBot3 are connected under the same IP.```
* Will try what you said tomorrow and let you know about the issue

ROBOTIS-Will, 2021-06-25:
@Prajwalmj 
Oh, my bad. You've done it correctly.
Let me see if I can reproduce the issue.

Prajwalmj, 2021-06-25:
@ROBOTIS-Will , Oh, okay, please let me know, what to do to solve this issue, as soon as possible. Thanks!

Prajwalmj, 2021-06-28:
@ROBOTIS-Will 
Just checking to see if there are any updates. Also, please let me know if there is anything else I need to do to overcome this issue.
Thanks for your time!

ROBOTIS-Will, 2021-06-29:
@Prajwalmj 
Thanks for your patience. 
I'll give you an update when there's any progress in reproducing the issue.

Prajwalmj, 2021-06-29:
@ROBOTIS-Will 
Thanks for your reply. Waiting for your update.

ROBOTIS-Will, 2021-07-01:
@Prajwalmj 
Just finished setting up mine and tried, but couldn't reproduce the problem.
[image]

If you are running other nodes (including roscore), please terminate them and try only running the command below on your Remote PC.
```
$ roslaunch turtlebot3_home_service_challenge_tools turtlebot3_pi_cam_remote.launch address:=192.168.0.153
```

Prajwalmj, 2021-07-01:
@ROBOTIS-Will 
* Thanks for your reply, I shutdown everything including roscore, turtlebot bringup, it works fine now. I think the ssh communication of home service challenge was getting interfered by the roscore and turtlebot ssh communication established previously.
* I have few questions, please let me know regarding this. As I said, I am trying to implement a simple pick and place task, with one source and destination in my own map. 
* So, other than editing room.yaml, scenario.yaml file, what else should I do, to use this code for my own task.
* I noticed that since in this challenge, there are 4 sources and 4 destinations, there 8 markers included in total, but I need only 2 for this case, where should I change this?
* Also, how to find the co-ordinates of marker position, as well as x,y co-ordinates as mentioned in room.yaml file, what I mean is how should I know the value of these co-ordinates if I know where it will be in the map.
* I am a beginner in ROS and I might have asked few silly questions as well. So, please let me know regarding this as soon as possible. Thanks you so much for your response

Prajwalmj, 2021-07-02:
@ROBOTIS-Will 
* So, I went further with SLAM, and saved my own map using this command
```ROS_NAMESPACE=tb3_hsc rosrun map_server map_saver -f tb3_hsc ```
* However, after that command, my map was saved in home directory.
* Then, I ran this command to look into a demo:
```roslaunch turtlebot3_home_service_challenge_tools turtlebot3_home_service_challenge_demo_remote.launch address:=192.168.0.153 ```
* However, rviz loaded with the map as in e-manual home service challenge instead of my own map.
* I looked into home service challenge package and noticed that maps are there in this path - ```turtlebot3_home_service_challenge/turtlebot3_home_service_challenge_tools/maps```
* So, I replaced the .pgm and .yaml file in there with my own map files saved using SLAM 
(I am guessing this is where the map should be?)
* Now, I run the demo using the command mentioned below, I notice that rviz gets loaded with my own map, but there is tf warning and almost all the plugins in the rviz have errors.
```roslaunch turtlebot3_home_service_challenge_tools turtlebot3_home_service_challenge_demo_remote.launch address:=192.168.0.153 ```
* I have attached the images to show the errors/warnings, please let me know what is the issue.
[image]
[image]

Prajwalmj, 2021-07-06:
Hi @ROBOTIS-Will,

Just checking in to see if you have any idea regarding this issue, as well as  how to get geometry pose 2D co-ordinates of room/destination for our own map.

ROBOTIS-Will, 2021-07-07:
Hi @Prajwalmj 
Have you tried setting the initial pose of TurtleBot3 with the `2D Pose Estimate` button on the top of RViz window?

Prajwalmj, 2021-07-07:
Hi @ROBOTIS-Will 
* My bad, what I meant when I asked about co-ordinates is, in room.yaml file, you have something like this. So, I'm assuming x, y are the range of co-ordinates for the room/target. So, for my own map, how do I get these values? I still don't understand this co-ordinate system, I was assuming you are using map-frame co-ordinate system here?
```room_1 :
  name : toilet
  object :
    marker : ar_marker_0
    position : [0.25, 0, 0.15]
  target :
    marker : ar_marker_4
    position : [0.25, 0, 0.15]
  x : [1.5, 0.6]
  y : [1.5, 0.2] 
```

* And regarding earlier tf transform error, this happens sometimes, while, sometimes I get the map correctly with green arrow marks and localization working, I make sure that every time, turtlebot and PC are connected  under same IP network. I'm still confused why this tf error is happening.
* 2D Pose Estimate, are you suggesting I can use that to get co-ordinates values? I thought that was just used for setting initial/starting location of robot in the map. This button works when that tf error is not there and all the localization and mapping features are working.
* Please let me know regarding this as soon as possible.

ROBOTIS-Will, 2021-07-08:
Hi @Prajwalmj 
Regarding the coordinate of the map, please refer to below issue thread.
https://github.com/ROBOTIS-GIT/turtlebot3_home_service_challenge/issues/5#issuecomment-631290064

In case of inconsistent behavior, it could be a network issue or [timesync](https://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#timesync-between-turtlebot3-and-remote-pc).
If tf is not received in the first place, 2D pose estimate won't work as well.

Prajwalmj, 2021-07-08:
Hi @ROBOTIS-Will,
* Thank you so much for that explanation in that thread, that cleared lots of doubts, I have one more question. From my own map, I have figured out how to get x and y coordinates range for the room, I still don't understand the position parameter in .yaml file and how to get this value for my own map?

``` room_1 :
name : toilet
object :
marker : ar_marker_0
position : [0.25, 0, 0.15] //Global coordinate of the object to grab with manipulator measured from the center of the marker.
target :
marker : ar_marker_4
position : [0.25, 0, 0.15] //Global coordinate where the object should be placed at.
x : [1.5, 0.6] //X coordinates of room 1 measured from the center of the arena.
y : [1.5, 0.2] //Y coordinates of room 1 measured from the center of the arena. With X and Y coordinates, when the robot moves to the room, calculate center coordinate of the room.
```
* I was planning to use publish point in rviz to get this 3d co-ordinate value, but don't understand what you mean by this "Global coordinate of the object to grab with manipulator measured from the center of the marker" for position parameter and how to measure this for my own map? 
* Regarding the timesync issue, so, the turtlebot3 and master PC is connected to the router that don't have internet connection. So, Pi cannot get current time and I have used chrony configuration to match time of pi with master PC. Before doing this, SLAM and other features where not working because of timesync issue and once I did that, everything started working. Do you think that could be the problem here for the tf error? Since I don't have the local network with internet connection, is there I can do if that is the reason for the tf error?

ROBOTIS-Will, 2021-07-09:
Hi @Prajwalmj 

- The position of object and target is the task space coordinate between the center of marker and the center of the object. However, it seems these values are not used but implemented directly in the code as below to grab objects from predetermined location.
https://github.com/ROBOTIS-GIT/turtlebot3_home_service_challenge/blob/master/turtlebot3_home_service_challenge_manager/src/tb3_hsc_manager.cpp#L445

- If you have correctly setup the chrony and synced time between two machines, you should be fine. I'm not sure why your tf timeout occurs. You can find similar cases in [ROS Answers](https://answers.ros.org/).

Prajwalmj, 2021-07-10:
Hi @ROBOTIS-Will,
* Now I think I can set those co-ordinates in the code directly according to my case then.
* I had two more questions. So, now if I have to test the entire thing, instead of testing individual commands, how should I run this, I mean what command I have to use after using below two commands, is it ready_mission and start_mission?
``` 
roslaunch turtlebot3_home_service_challenge_tools turtlebot3_home_service_challenge_demo_remote.launch address:=192.168.0.153
roslaunch turtlebot3_home_service_challenge_manager manager.launch
```
* Regarding the tf error, I noticed one thing recently, whenever I get that tf error, I notice that the LIDAR is not scanning, I mean not spinning. I am assuming that this might be because bringup is not being launched properly(correct me if I am wrong)I don't why this command is unable to do the bringup sometimes, do you have any idea?
```
 roslaunch turtlebot3_home_service_challenge_tools turtlebot3_home_service_challenge_demo_remote.launch address:=192.168.0.153 
```
* Besides I can't do turtlebot3 bringup separately because to run this home_service_challenge_package, I have to kill all other nodes and terminals, otherwise I get that "remote roslaunch failed to launch:tb3" error. I thought the command above should be able to do the bringup as well as mentioned in the lab manual.

ROBOTIS-Will, 2021-07-12:
@Prajwalmj 
The below steps will launch the official HSC on an official map.

1. Bringup TurtleBot3 and launch other necessary nodes such as camera, manipulation, and navigation nodes
```
$ roslaunch turtlebot3_home_service_challenge_tools turtlebot3_home_service_challenge_demo_remote.launch address:=TURTLEBOT_IP
```
2. Launch the challege manager
```
$ roslaunch turtlebot3_home_service_challenge_manager manager.launch
```
3. Mission control
```
### ​ready and start the mission
$ rostopic pub -1 /tb3_hsc/command std_msgs/String ready_mission
$ rostopic pub -1 /tb3_hsc/command std_msgs/String start_mission

### stop the mission
$ rostopic pub -1 /tb3_hsc/command std_msgs/String stop_mission
```

If the Lidar is not rotating, your bringup is not successful. Please make sure that all the connections are securely made.
When launching the `turtlebot3_home_service_challenge_demo_remote.launch` file, the LDS should be running.

Since [bringup is already included in the above launch file](https://github.com/ROBOTIS-GIT/turtlebot3_home_service_challenge/blob/master/turtlebot3_home_service_challenge_tools/launch/turtlebot3_home_service_challenge_demo_remote.launch#L6), you don't need to launch a separate bringup.

I'll transfer this issue thread to [https://github.com/ROBOTIS-GIT/turtlebot3_home_service_challenge/issues](https://github.com/ROBOTIS-GIT/turtlebot3_home_service_challenge/issues) after your next reply.

Thanks!

Prajwalmj, 2021-07-12:
Hi @ROBOTIS-Will,

* I had few more questions, I tried using the mission control commands, however it's not working for some reason, turtlebot3 goes to the object position in the map, then tries to push the block holding object instead of stopping at the location. I made sure that when I test individual commands, especially the commands below, turtlebot3 perfectly stops near markers 0.
```
rostopic pub -1 /tb3_hsc/command std_msgs/String nav_start
rostopic pub -1 /tb3_hsc/command std_msgs/String nav_ar_marker_0
```
[image]
[image]
 * I have attached map details as shown above, I will put the command results below in the zip file as the log results are too long.
[Github_issue.zip](https://github.com/ROBOTIS-GIT/open_manipulator/files/6803568/Github_issue.zip)

* Please let me know how to overcome this issue

ROBOTIS-Will, 2021-07-13:
In the first image of the reply above, I can see that the AR marker is located outside of the map.
Make sure that you are using the correct size of the marker (90 x 90 mm).
Also, it is recommended to use the `teleop keyboard` to get more focused amcl particles(in green arrows around the TurtleBot3)
[image]  [image]

Prajwalmj, 2021-07-13:
Hi @ROBOTIS-Will,

* I was earlier using remote controller for SLAM navigation to create map, but now, I did use teleop to create map again.
* Also, I changed the marker picture to 90x90 mm size, however, I still get the AR marker out of the map for some reason I don't understand. Does other things like height of the box on which marker is pasted or height at which marker is pasted matter? As, you can see from this image, the marker is clearly in camera's vision, but in map it's still showing it outside the map.
[image]

* Please let me know why this is happening

ROBOTIS-Will, 2021-07-14:
@Prajwalmj 
I'm not quite sure why your AR marker is located at the outside of the map.
What I meant for using keyboard teleop is to concentrate the amcl particles by rotating or moving the TurtleBot3 within the map.
This will help increasing the accuracy of TurtleBot3 pose during navigation.

Prajwalmj, 2021-07-14:
@ROBOTIS-Will 
* I understand, I think I will try to concatenate more amcl particles within the map and see if that will affect AR marker position.
* If that doesn't work, I will create a new map and see if the situation same for the new map as well! will get back to you in couple of days. Thanks!

ROBOTIS-Will, 2021-07-14:
@Prajwalmj 
Thanks for your understandings.
I'll also try the AR marker detection part, but I might take a few weeks to test this.
I'll update you once I finish testing with my Waffle Pi.

yun-goon, 2024-12-19:
This issue appears to be resolved, so it will be closed.

closed 2024-12-19 by yun-goon