ROBOTIS-GIT/open_manipulator #213: Unable to lower minimum on Gripper Close past -0.10

sardanian, 2021-07-04:
I am trying to make the gripper close just a little bit more. It is currently set to a minimum of -0.10, but I want to close it just a bit more. I can manipulate the gripper to close a bit more, but cannot seem to find in the code what to change. I have attempted to change it in the:
open_manipulator_x_teleop_keyboard.cpp file (line 239) but it stops working when it is put to anything below -0.10

I then assumed it had a minimum and tried to change it in: open_manipulator_x.cpp (line 114) but that also doesn't work.

Is there a way I can change the code so that the gripper can close a little bit more?

Thank you.

ROBOTIS-Will, 2021-07-05:
@sardanian 
Here's the configuration of each joint of OM-X and you can find the minimum gripper value in the below.
https://github.com/ROBOTIS-GIT/open_manipulator/blob/master/open_manipulator_libs/src/open_manipulator.cpp#L114

Thanks!

sardanian, 2021-07-05:
Hi @ROBOTIS-Will,

I had previously tried to change that file and it would not work.

What I changed was the value from -.01 to -.02. I also changed this line to -.02 as well.

https://github.com/ROBOTIS-GIT/open_manipulator/blob/master/open_manipulator_teleop/src/open_manipulator_teleop_keyboard.cpp#L328

The arm will not close when the values of both files are below -.01

Am I missing something?

Thank you.

ROBOTIS-Will, 2021-07-06:
@sardanian 
The gripper of OM-X is mechanically limited by its design.
Make sure that there's a room for exceeding the minimum pose.
I believe -0.01 is pretty close to the mechanical limit and in order to grab something thin, you should use thicker gripper pad that is included in the package.
[image]

sardanian, 2021-07-07:
Hi Will. I have more mechanical space.

This is the most my gripper will close via software.

However, this is what it can mechanically close to if I move it with my
hand when turned off. So it seems it can close 5mm more, but I cannot get
it to close via the software. When I change the lines of code you suggested
the gripper just will not close at all. It stays stuck in the up position.
So it isn’t a mechanical issue. It seems to stop and then not work/close at
all.

ROBOTIS-Will, 2021-07-08:
Hi @sardanian 
The pictures are not attached in your reply.
Modifying below two lines with a value -0.014 allowed me to fully close the gripper to its mechanical minimum.
- https://github.com/ROBOTIS-GIT/open_manipulator/blob/master/open_manipulator_libs/src/open_manipulator.cpp#L114
- https://github.com/ROBOTIS-GIT/open_manipulator/blob/master/open_manipulator_teleop/src/open_manipulator_teleop_keyboard.cpp#L328

closed 2021-08-06 by sardanian