ROBOTIS-GIT/open_manipulator #216: Unable to launch pick and place example using Gazebo

cybernetchi, 2021-07-22:
Hi,

Ubuntu 16.04
ROS Kinetic

I am trying to follow https://emanual.robotis.com/docs/en/platform/openmanipulator_x/ros_perceptions/#simulation
to launch the pick and place using Gazebo.

- When I am at this step
  `roslaunch open_manipulator_ar_markers ar_pose.launch camera_model:=realsense_d435 use_platform:=false`
  the robot model is not properly loaded

[image]

- There is also no image received from the camera

- Therefore I can't reach open_manipulator_pick_and_place.launch

Thank you

ROBOTIS-Will, 2021-07-23:
Hi @cybernetchi 
It looks like undeveloped test code had been updated to the manual.
AR simulation is currently not supported, but I'll review this after the update that I'm currently working on.
Thanks!

yun-goon, 2024-12-19:
Apologies for the delayed response. We have assembled a TF team to address OpenManipulator-x issues and are actively working on resolutions. Please note that we do not support EOL (End-of-Life) ROS versions; our support currently extends to Noetic and Humble versions. We kindly ask that you try again with the latest version. If the same issue arises, please feel free to resubmit it, and we will address it promptly.

closed 2024-12-19 by yun-goon