ROBOTIS-GIT/open_manipulator #217: 6DOF OpenManipulator

cybernetchi, 2021-08-12:
Hi, when I catkin_make the following:

open_manipulator_friends.git
open_manipulator_6dof_simulations.git
open_manipulator_6dof_application.git

I got the compile error:

fatal error: open_manipulator_motion/MotionState.h: No such file or directory
   52 | #include "open_manipulator_motion/MotionState.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

Would greatly appreciate if you could give me some help?
Thank you very much

ROBOTIS-Will, 2021-08-17:
Hi @cybernetchi 
The example is supported in ROS1 Kinetic.
The `MotionState.h` is created during source build and placed under the ***catkin_ws/devel/include/open_manipulator_motion***
You may try using a single thread when building the source as below.
```bash
$ catkin_make -j1
```

cybernetchi, 2021-08-17:
Hi @ROBOTIS-Will
Thank you for your reply.

Yes I am in ROS1 Kinetic and tried building with single thread but got the same result:

> [ 77%] Generating qrc_images.cpp
> [ 77%] Generating include/open_manipulator_6dof_control_gui/moc_main_window.cpp
> [ 77%] Generating include/open_manipulator_6dof_control_gui/moc_qnode.cpp
> [ 78%] Generating /home/cybernetchi/catkin_ws/devel/include/open_manipulator_6dof_control_gui/ui_main_window.h
> Scanning dependencies of target open_manipulator_6dof_control_gui
> [ 78%] Building CXX object open_manipulator_6dof_application/open_manipulator_6dof_control_gui/CMakeFiles/open_manipulator_6dof_control_gui.dir/src/main.cpp.o
> In file included from /home/cybernetchi/catkin_ws/src/open_manipulator_6dof_application/open_manipulator_6dof_control_gui/src/../include/open_manipulator_6dof_control_gui/main_window.hpp:32:0,
>                  from /home/cybernetchi/catkin_ws/src/open_manipulator_6dof_application/open_manipulator_6dof_control_gui/src/main.cpp:29:
> /home/cybernetchi/catkin_ws/src/open_manipulator_6dof_application/open_manipulator_6dof_control_gui/src/../include/open_manipulator_6dof_control_gui/qnode.hpp:52:49: fatal error: open_manipulator_motion/MotionState.h: No such file or directory
> compilation terminated.
> open_manipulator_6dof_application/open_manipulator_6dof_control_gui/CMakeFiles/open_manipulator_6dof_control_gui.dir/build.make:79: recipe for target 'open_manipulator_6dof_application/open_manipulator_6dof_control_gui/CMakeFiles/open_manipulator_6dof_control_gui.dir/src/main.cpp.o' failed
> make[2]: *** [open_manipulator_6dof_application/open_manipulator_6dof_control_gui/CMakeFiles/open_manipulator_6dof_control_gui.dir/src/main.cpp.o] Error 1
> CMakeFiles/Makefile2:5764: recipe for target 'open_manipulator_6dof_application/open_manipulator_6dof_control_gui/CMakeFiles/open_manipulator_6dof_control_gui.dir/all' failed
> make[1]: *** [open_manipulator_6dof_application/open_manipulator_6dof_control_gui/CMakeFiles/open_manipulator_6dof_control_gui.dir/all] Error 2
> Makefile:138: recipe for target 'all' failed
> make: *** [all] Error 2
> Invoking "make -j1" failed

I can see the folder **_catkin_ws/devel/include/open_manipulator_motion_** was created but there is no `MotionState.h`.
Thank you very much!

ROBOTIS-Will, 2021-08-30:
@cybernetchi 
Sorry about the delayed response.
Could you try rebuilding the code after deleting devel and build directories in the catkin_ws folder?

cybernetchi, 2021-08-30:
Hi @ROBOTIS-Will 
Thank you for your reply.

No luck even if I deleted  `devel `and `build `directories and `catkin_make` again.
It comes the same error. Any idea why? :/
Thank you!

ROBOTIS-Will, 2021-09-06:
Hi @cybernetchi 

I've tested on my machine, but wasn't able to reproduce the issue.
Reinstalling ROS might help to resolve the issue.
Thanks!

yun-goon, 2024-12-19:
This issue appears to be resolved, so it will be closed.

closed 2024-12-19 by yun-goon