ROBOTIS-GIT/open_manipulator #218: Using openmanipulator with an AX-12A

Lantc26, 2021-08-16:
Hello,
we would like to use openmanipulator with an [AX-12A](https://emanual.robotis.com/docs/en/dxl/ax/ax-12a/).
We already saw [this](https://github.com/ROBOTIS-GIT/open_manipulator/issues/68) post claiming, that no AX-12 servos are supported.
We'd like to modify the code to support our robot.
We already followed [these advices](https://github.com/ROBOTIS-GIT/open_manipulator/issues/84#issuecomment-459623525) and modified the code to match our robot.
At the moment we get this error:

```[ERROR] [TxRxResult] There is no status packet!
[ERROR] Please check your Dynamixel ID
[ERROR] [TxRxResult] There is no status packet!
[ERROR] Please check your Dynamixel ID
[ERROR] [TxRxResult] There is no status packet!
[ERROR] Please check your Dynamixel ID
[ERROR] [TxRxResult] There is no status packet!
[ERROR] Please check your Dynamixel ID
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [TxRxResult] There is no status packet!
[ERROR] Please check your Dynamixel ID
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelWorkbench] Failed to change torque status!
[ERROR] [DynamixelWorkbench] Failed to change torque status!
[ERROR] [DynamixelWorkbench] Failed to change torque status!
[ERROR] [DynamixelWorkbench] Failed to change torque status!
[ERROR] [DynamixelWorkbench] Failed to change torque status!
[ERROR] [TxRxResult] There is no status packet!
[ERROR] groupSyncRead getdata failed
[ERROR] groupSyncRead getdata failed
[ERROR] groupSyncRead getdata failed
```
This has probably to do with the unsupported AX12 servo.
Thank you in advance

ROBOTIS-Will, 2021-08-17:
Hi @Lantc26 
Please understand that we do not provide a thorough technical support for customizing the product.
Since open_manipulator packages are written for X series which runs off of DYNAMIXEL Procotol 2.0, converting them to support Protocol 1.0 will require a moderate level of rewriting source codes.
Thank you.

Lantc26, 2021-09-13:
I see. We found a way to control the arm in a different way. So I close this issue now.
Thank you anyway.

closed 2021-09-13 by Lantc26

alex-ssom, 2021-09-28:
Hi @Lantc26,

I'm having the same issue. Would you mind sharing the solution you used?