ROBOTIS-GIT/open_manipulator #219: Using open_manipulator_6dof with Moveit!

cybernetchi, 2021-09-02:
Hi,

ROS Kinectic on Ubuntu 16.06
Joint Dynamixel ID : 11, Model Name : XM430-W210
Joint Dynamixel ID : 12, Model Name : XH430-W350
Joint Dynamixel ID : 13, Model Name : XH430-W350
Joint Dynamixel ID : 14, Model Name : XM430-W210
Joint Dynamixel ID : 15, Model Name : XM430-W210
Joint Dynamixel ID : 16, Model Name : XM430-W210
Gripper Dynamixel ID : 17, Model Name :XM430-W210

I am following [this](https://github.com/zang09/open_manipulator_6dof_application) and made a 6dof open manipulator.
When I use the moveit, the arm_group works fine.
However, I am not able to control the gripper with MoveIt Commander Scripting, even if the range is -0.01 ~ 0.01.
I can see the path planned in Rviz, but it is not executed to the real robot.

This is the error:
```
[INFO] [OpenManipulator Controller] Execute Moveit planned path
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 1) >= this->size() (which is 1)
[open_manipulator_6dof-6] process has died [pid 41350, exit code -6, cmd /home/cybernetchi/catkin_ws/devel/lib/open_manipulator_6dof_controller/open_manipulator_6dof_controller /dev/ttyUSB0 1000000 __name:=open_manipulator_6dof __log:=/home/cybernetchi/.ros/log/518d038e-0c2c-11ec-baa5-000c2960058a/open_manipulator_6dof-6.log].
log file: /home/cybernetchi/.ros/log/518d038e-0c2c-11ec-baa5-000c2960058a/open_manipulator_6dof-6*.log
```
Any help would be appreciated!
Thank you very much!

ROBOTIS-Will, 2021-09-06:
Hi @cybernetchi 

The 6 DOF OpenMANIPULATOR is created by a community developer so please contact the developer for more details.
If I were you, I'd run the keyboard teleop and check the gripper's working range.
https://github.com/zang09/open_manipulator_6dof_application#keyboard

Thank you.

yun-goon, 2024-12-19:
This issue will be closed due to prolonged inactivity.

closed 2024-12-19 by yun-goon