ROBOTIS-GIT/open_manipulator #222: open_manipulator keyboard freezes

ROBOTIS-Will, 2021-11-16:
This issue was reproduced in the environment below.
- Ubuntu 18.04
- ROS Melodic
- OpenMANIPULATOR-X (default settings)
- U2D2 (port latency 1ms)

How to reproduce this issue
1. Launch the controller
```
$ roslaunch open_manipulator_controller open_manipulator_controller.launch 
```
2. Launch the teleop node and send keyboard commands continuously.
```
$ roslaunch open_manipulator_teleop open_manipulator_teleop_keyboard.launch
```

DarrylHsu0105, 2022-01-17:
Hi, I am currently facing the same problem.
Ubuntu 20.04.3 LTS
OpenMANIPULATOR-P (default settings)
U2D2 (port latency 1ms)

And I used same command with you.
Now , I can't using GUI or Keyboard teleop control it.
I can only use "Actuator" and "read" button , everything else causes a freeze.
So, how can fix it. thank you.
By the way, I'm sure all of my joint are all within the working range (1.571 rad).

ROBOTIS-Will, 2022-01-25:
Hi @DarrylHsu0105 
In case of operating the OpenMANIPULATOR-P, you should use the openmanipulator-p package.
https://github.com/ROBOTIS-GIT/open_manipulator_p
Which ROS version do you use?

JYKimm, 2023-11-22:
not fixed yet?

ChangAnCN, 2024-04-23:
Same issue. can not use it.

gholibqasobov, 2024-05-18:
HI there, has anyone solved it? I am having the same issue.

sunghowoo, 2025-01-13:
This issue appears to be resolved, so it will be closed.

closed 2025-01-13 by sunghowoo