ROBOTIS-GIT/open_manipulator #223: How can I controlled Open Manipulator X with XM430-W350-R ?

ahmetmustafakos, 2021-11-25:
I am following this tutorial : https://emanual.robotis.com/docs/en/platform/openmanipulator_x/quick_start_guide/
My motors communicates with RS485 as oppose to TLL. 
When I upload the OpenCR  with arduino IDE using example code "open_manipulator_chain" and then when I try to control manipulator with Processing I get an error saying;
 "Please, set on controller 
[0] "COM5"
[1] "COM6"
[2] "COM7"

My openCR is connetted to COM7 I set this setting  like this: " connectOpenCR(2); "

What should I do to control this manipulator?

ROBOTIS-Will, 2021-11-29:
Hi @ahmetmustafakos 

Some latest version of Processing may not compatible to the instructions.
Please try using the Processing v3.5.4.

Also, make sure to press the reset button of the OpenCR before running the processing as written in the eManual below.
https://emanual.robotis.com/docs/en/platform/openmanipulator_x/quick_start_guide_basic_operation/#launch-controller

When the Processing is up and running, you should click on the `CONTROLLER ON` button first then manipulate the arm with the GUI.
[image]

ahmetmustafakos, 2021-12-02:
hello @ROBOTIS-Will
I tested using Processing v3.5.4 version. I can make changes in the GUI without any errors. But there is no movement in my motors. Could this tutorial be written for the XM430-W350-T motors communicating with TTL? Because I am using the XM430-W350-R model motors. This motors uses RS485. Do I need to make changes to the code so I can use these motors?

ROBOTIS-Will, 2021-12-03:
Hi @ahmetmustafakos 

When resetting the OpenCR, do DYNAMIXEL gets powered(LED blinks once) and torque on (stiff joint)?
The example code is written for the official OpenMANIPULATOR-X package and you don't need to change anything.
As long as you have correctly connected DYNAMIXEL to the RS485 port of OpenCR, it doesn't matter whether you are running TTL or RS485 DYNAMIXEL.
Thanks.

ahmetmustafakos, 2021-12-03:
Hi @ROBOTIS-Will 
When I reset OpenCR, the motors leds are blinking but the motors are not getting torque.

ROBOTIS-Will, 2021-12-06:
@ahmetmustafakos 

If DYNAMIXEL modules are not getting the torque after pressing the OpenCR reset button, most likely your DYNAMIXEL have not been configured correctly.
The OpenMANIPULATOR-X package comes with DYNAMIXEL configured as below.
- ID : 11 (Base yaw joint) ~ 15 (Gripper joint)
- Baudrate : 1Mbps

You should match the default setting in order to run the Arduino example.
Thanks!

yun-goon, 2024-12-19:
This issue will be closed due to prolonged inactivity.

closed 2024-12-19 by yun-goon