ROBOTIS-GIT/open_manipulator #224: Shut down actuators from python code

auxoche, 2022-01-17:
Good evening, is it possible to access low level commands for the actuators easily from Python code? I noticed that the open_manipulator_controller turns off the motors before exiting while a code with MoveIt will leave them on.

Are there any python code examples that address this? Would also accept python code examples that use the default open_manipulator_controller instead.

ROBOTIS-Will, 2022-01-25:
Hi @auxoche 
You may comment the below code out in order to keep the torque of the DYNAMIXEL when the node is destroyed.
https://github.com/ROBOTIS-GIT/open_manipulator/blob/ros2/open_manipulator_x_controller/src/open_manipulator_x_controller.cpp#L57

yun-goon, 2024-12-19:
This issue will be closed due to prolonged inactivity. Please try again with the Humble or Noetic versions, and if the issue persists, feel free to reopen it.

closed 2024-12-19 by yun-goon