ROBOTIS-GIT/open_manipulator #225: How to tune the friction coefficient of OpenMANIPULATOR's gripper

Yoshihito-M, 2022-01-25:
Hi,everyone. 
I am using TurtleBot3 Waffle Pi with OpenMANIPULATOR-X.
Now, I am trying the task gripping a cylinder model with the gripper of OpenMANIPULATOR-X on gazebo.
however, the gripper model can not grip the cylinder model.
Namely, the cylinder model  slips off the surface of gripper.

I guess that the friction coefficient of the surface of gripper should be tuned, but I'm not sure what process I should do overall.
I would appreciate it if you could help me tune it.

Thanks for your help.

ROBOTIS-Will, 2022-01-25:
Hi @Yoshihito-M 

This is more relevant to the Gazebo simulator.
The following thread might help, but can't assure whether these approaches will work or not.
https://answers.gazebosim.org//question/22215/friction-in-the-gripper-not-working-objects-slipping/

Thanks!

Yoshihito-M, 2022-01-25:
Hi @ROBOTIS-Will 

Thank you for showing the helpful thread.
I fixed open_manipulator_x.gazebo.xacro as follows:
-------------------------------------------------------------------
  
      <gazebo reference="gripper_link_sub">
      <kp>1000000.0</kp>
      <kd>100000.0</kd>
      <mu1>1000.0</mu1>
      <mu2>1000.0</mu2>
      <maxVel>0.0</maxVel>
      <minDepth>0.001</minDepth>
      <material>Gazebo/DarkGrey</material>
    </gazebo>
----------------------------------------------------------------

Yoshihito-M, 2022-01-26:
Hi @ROBOTIS-Will

I tried  above code on Gazebo Simulator and I solved my issue!
your advice is very helpful for me. Thank you again!

closed 2022-01-26 by Yoshihito-M

ROBOTIS-Will, 2022-01-26:
@Yoshihito-M 
Thanks for the updates!
I'll refer to your case when updating the xacro file in the next release.