ROBOTIS-GIT/open_manipulator #226: How to operate OpenMANIPULATOR in both Real Environment and Gazebo World

Yoshihito-M, 2022-02-01:
Hi,everyone.
I am using TurtleBot3 Waffle Pi with OpenMANIPULATOR-X.
Now, I am trying the task gripping a cylinder with the gripper of OpenMANIPULATOR-X .
In addition, I would like to simultaneously operate  OpenMANIPULATOR-X both in real world and that model in Gazebo world.
I could find helpful launch file to achieve my task in the below link, but the file show how to operate it not in Gazebo world but in **Rviz** world.
https://emanual.robotis.com/docs/en/platform/turtlebot3/manipulation/#operate-the-actual-openmanipulator

I would appreciate it if you could help me operate it in Gazebo world.
Thanks for your help.

Yoshihito-M, 2022-02-02:
@ROBOTIS-Shibata 

Thanks for your advice.
Can I download these document and launch files from Github?

ROBOTIS-Shibata, 2022-02-03:
@Yoshihito-M 

日本の方とお見受けします。
ROBOTIS日本支店で個別の対応をしますので、お手数ですが以下のメアドにコンタクトをお願いします。
jp.support@robotis.com

Yoshihito-M, 2022-02-04:
@ROBOTIS-Shibata 
お世話になっております。
上記拝承しました。
ご連絡します。

closed 2022-02-04 by Yoshihito-M