ROBOTIS-GIT/open_manipulator #228: Gazebo for Ros2?

sardanian, 2022-03-15:
When is Gazebo for Ros2 supposed to be implemented? The emanual mentions Q4 2021.

Also, is there a timeframe for MoveIt,  Gravity Compensation and GUI? The emanual says "not supported yet" which makes me think it is on the way but not there yet.

EndlessLoops, 2022-11-03:
Same question.

yun-goon, 2024-12-19:
@sardanian Apologies for the delayed response. We have assembled a TF team to address OpenManipulator-x issues and are actively working on resolutions. Please note that we do not support EOL (End-of-Life) ROS versions; our support currently extends to Noetic and Humble versions. We kindly ask that you try again with the latest version. If the same issue arises, please feel free to resubmit it, and we will address it promptly.

closed 2024-12-19 by yun-goon