ROBOTIS-GIT/open_manipulator #229: Quite a few typos or information not relavent to Ros2

sardanian, 2022-03-15:


ROBOTIS-David, 2022-03-15:
Thank you for raising an issue regarding the typo and some misleading content, I will have our engineers check it and make a change on your request.  

* Please allow me to transfer to the proper repos. This issue belongs to another branch, OpenManipulator-X.

ashekim, 2022-03-21:
Hi @sardanian. 
This content will be corrected and reflected in the e-manual soon.
Thank you for your contribution!

@ROBOTIS-David 
Please update the e-manual after checking the revisions.

ROBOTIS-David, 2022-03-21:
Thank you for checking this out.
The given PR is confirmed and has been merged to the Master branch of e-Manual . Please close the ticket if solved.

yun-goon, 2024-12-19:
This issue appears to be resolved, so it will be closed.

closed 2024-12-19 by yun-goon