ROBOTIS-GIT/open_manipulator #233: unable to build packages

sardanian, 2022-06-25:
I for some reason cannot run a colcon build without errors anymore.

I have removed the workspace and redownloaded all sourcecode again.

This is the errors I am getting:

Starting >>> dynamixel_sdk
Starting >>> cmake_modules
Starting >>> open_manipulator_msgs
Starting >>> dynamixel_sdk_custom_interfaces
Starting >>> open_manipulator_x_description
Finished <<< cmake_modules [0.08s]
Starting >>> robotis_manipulator
Finished <<< open_manipulator_x_description [0.09s]                                                                                                                                                          
Finished <<< dynamixel_sdk [0.12s]
Starting >>> dynamixel_workbench_toolbox
--- stderr: open_manipulator_msgs
make[1]: *** No rule to make target 'open_manipulator_msgs__py/CMakeFiles/open_manipulator_msgs__py.dir/all', needed by 'CMakeFiles/open_manipulator_msgs__python.dir/all'.  Stop.
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:146: all] Error 2
---
Failed   <<< open_manipulator_msgs [0.13s, exited with code 2]
Aborted  <<< dynamixel_workbench_toolbox [0.28s]                                                                                                                                   
Aborted  <<< dynamixel_sdk_custom_interfaces [0.50s]                                                                               
Aborted  <<< robotis_manipulator [4.35s]                                 

Summary: 3 packages finished [4.57s]
  1 package failed: open_manipulator_msgs
  3 packages aborted: dynamixel_sdk_custom_interfaces dynamixel_workbench_toolbox robotis_manipulator
  2 packages had stderr output: dynamixel_workbench_toolbox open_manipulator_msgs
  6 packages not processed

Any help would be greatly appreciated.

sardanian, 2022-06-25:
I also get this error:

Starting >>> dynamixel_sdk
Starting >>> cmake_modules
Starting >>> open_manipulator_msgs
Starting >>> dynamixel_sdk_custom_interfaces
Starting >>> open_manipulator_x_description
Finished <<< cmake_modules [0.25s]                                                                                                                                                  
Starting >>> robotis_manipulator
Finished <<< open_manipulator_x_description [0.25s]
Finished <<< dynamixel_sdk [0.28s]
Starting >>> dynamixel_workbench_toolbox
--- stderr: dynamixel_sdk_custom_interfaces                                                                                                                                                         
make[1]: *** No rule to make target 'dynamixel_sdk_custom_interfaces__py/CMakeFiles/dynamixel_sdk_custom_interfaces__py.dir/all', needed by 'CMakeFiles/dynamixel_sdk_custom_interfaces__python.dir/all'.  Stop.
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:146: all] Error 2
---
Failed   <<< dynamixel_sdk_custom_interfaces [0.55s, exited with code 2]
Aborted  <<< robotis_manipulator [0.37s]                                                                                                                        
Aborted  <<< dynamixel_workbench_toolbox [0.37s]
Aborted  <<< open_manipulator_msgs [0.74s]                            

Summary: 3 packages finished [0.88s]
  1 package failed: dynamixel_sdk_custom_interfaces
  3 packages aborted: dynamixel_workbench_toolbox open_manipulator_msgs robotis_manipulator
  1 package had stderr output: dynamixel_sdk_custom_interfaces
  6 packages not processed

yun-goon, 2024-12-19:
Apologies for the delayed response. We have assembled a TF team to address OpenManipulator-x issues and are actively working on resolutions. Please note that we do not support EOL (End-of-Life) ROS versions; our support currently extends to Noetic and Humble versions. We kindly ask that you try again with the latest version. If the same issue arises, please feel free to resubmit it, and we will address it promptly.

closed 2024-12-19 by yun-goon