ROBOTIS-GIT/open_manipulator #235: open manipulator x unable to connect to port error

gandrade2, 2022-07-07:
[image]

I am getting this error and am unsure what to do. ROS was installed correctly, and I am able to manipulate the arm using dynamixel_wizard2

ROBOTIS-Will, 2022-07-21:
Hi @gandrade2 

If you are using U2D2 as USB-DYNAMIXEL interface, make sure to use the command below without any options
```
$ roslaunch open_manipulator_controller open_manipulator_controller.launch
```

If you are using OpenCR as an interface, please upload the `usb_dxl` Arduino sketch to the OpenCR.
https://emanual.robotis.com/docs/en/platform/openmanipulator_x/quick_start_guide/#communication-interface

Click on the `Option2` box to unfold the content.
[image]

Thanks!

mq036, 2023-12-18:
@ROBOTIS-Will I am having the same issue and i am using U2D2 board for the control with the above-mentioned command and still it shows the same error. I have also tried changing the USB connector cable but no luck.

yun-goon, 2024-12-19:
Apologies for the delayed response. We have assembled a TF team to address OpenManipulator-x issues and are actively working on resolutions. Please note that we do not support EOL (End-of-Life) ROS versions; our support currently extends to Noetic and Humble versions. We kindly ask that you try again with the latest version. If the same issue arises, please feel free to resubmit it, and we will address it promptly.

closed 2024-12-19 by yun-goon