ROBOTIS-GIT/open_manipulator #236: Openmanipulator There is no status packet, Failed to get the tool

padre221, 2022-08-05:
I have 4 XM430-W350-R motors setup on openmanipulator with no gripper.
-OS UBUNTU 16.04 LTS
-ROS Kinetic
-OpenCR
-My robot is connected to OpenCR board

I have been following the instructions on the openmanipulator-x manual ([OpenMANIPULATOR-X 1](https://emanual.robotis.com/docs/en/platform/openmanipulator_x/quick_start_guide/#setup)) webpage and i have done the instructions on step 4. As you can understand i am trying to control the robot via openCR and ROS. After the step 4 Option 2 : Use OpenCR as a communication interface instructions i proceed to 4.2.Basic Operation tab which is telling me to run this command ;

`user@user:~/catkin_ws$ roslaunch open_manipulator_controller open_manipulator_controller.launch usb_port:=/dev/ttyACM0 baud_rate:=1000000`

and when i run it i get the below result;

`user@user:~/catkin_ws$ roslaunch open_manipulator_controller open_manipulator_controller.launch usb_port:=/dev/ttyACM1 baud_rate:=1000000
... logging to /home/user/.ros/log/b791810a-1400-11ed-bab2-507b9d65ae7d/roslaunch-user-2029.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://user:33851/

SUMMARY
========

PARAMETERS
 * /open_manipulator/control_period: 0.01
 * /open_manipulator/moveit_sample_duration: 0.05
 * /open_manipulator/planning_group_name: arm
 * /open_manipulator/using_moveit: False
 * /open_manipulator/using_platform: True
 * /rosdistro: kinetic
 * /rosversion: 1.12.17

NODES
  /
    open_manipulator (open_manipulator_controller/open_manipulator_controller)

ROS_MASTER_URI=http://localhost:11311

process[open_manipulator-1]: started with pid [2052]
Joint Dynamixel ID : 11, Model Name : XM430-W350
Joint Dynamixel ID : 12, Model Name : XM430-W350
Joint Dynamixel ID : 13, Model Name : XM430-W350
Joint Dynamixel ID : 14, Model Name : XM430-W350
[ERROR] [TxRxResult] There is no status packet!
[ERROR] Please check your Dynamixel ID
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelDriver] Failed to get the Tool
[ERROR] [DynamixelWorkbench] Failed to change torque status!
[open_manipulator-1] process has died [pid 2052, exit code -11, cmd /home/user/catkin_ws/devel/lib/open_manipulator_controller/open_manipulator_controller /dev/ttyACM0 1000000 __name:=open_manipulator __log:=/home/user/.ros/log/b791810a-1400-11ed-bab2-507b9d65ae7d/open_manipulator-1.log].
log file: /home/user/.ros/log/b791810a-1400-11ed-bab2-507b9d65ae7d/open_manipulator-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done`

Here is my launch file 

`<launch>
  <arg name="use_robot_name"         default="open_manipulator"/>

  <arg name="dynamixel_usb_port"     default="/dev/ttyACM0"/>
  <arg name="dynamixel_baud_rate"    default="1000000"/>

  <arg name="control_period"         default="0.010"/>

  <arg name="use_platform"           default="true"/>

  <arg name="use_moveit"             default="false"/>
  <arg name="planning_group_name"    default="arm"/>
  <arg name="moveit_sample_duration" default="0.050"/>

  <group if="$(arg use_moveit)">
    <include file="$(find open_manipulator_controller)/launch/open_manipulator_moveit.launch">
      <arg name="robot_name"      value="$(arg use_robot_name)"/>
      <arg name="sample_duration" value="$(arg moveit_sample_duration)"/>
    </include>
  </group>

  <node name="$(arg use_robot_name)" pkg="open_manipulator_controller" type="open_manipulator_controller" output="screen" args="$(arg dynamixel_usb_port) $(arg dynamixel_baud_rate)">
      <param name="using_platform"       value="$(arg use_platform)"/>
      <param name="using_moveit"         value="$(arg use_moveit)"/>
      <param name="planning_group_name"  value="$(arg planning_group_name)"/>
      <param name="control_period"       value="$(arg control_period)"/>
      <param name="moveit_sample_duration"  value="$(arg moveit_sample_duration)"/>
  </node>

</launch>`

padre221, 2022-08-05:
I could not put my launch file for some reason so here it is 
<launch>
  <arg name="use_robot_name"         default="open_manipulator"/>

  <arg name="dynamixel_usb_port"     default="/dev/ttyACM0"/>
  <arg name="dynamixel_baud_rate"    default="1000000"/>

  <arg name="control_period"         default="0.010"/>

  <arg name="use_platform"           default="true"/>

  <arg name="use_moveit"             default="false"/>
  <arg name="planning_group_name"    default="arm"/>
  <arg name="moveit_sample_duration" default="0.050"/>

  <group if="$(arg use_moveit)">
    <include file="$(find open_manipulator_controller)/launch/open_manipulator_moveit.launch">
      <arg name="robot_name"      value="$(arg use_robot_name)"/>
      <arg name="sample_duration" value="$(arg moveit_sample_duration)"/>
    </include>
  </group>

  <node name="$(arg use_robot_name)" pkg="open_manipulator_controller" type="open_manipulator_controller" output="screen" args="$(arg dynamixel_usb_port) $(arg dynamixel_baud_rate)">
      <param name="using_platform"       value="$(arg use_platform)"/>
      <param name="using_moveit"         value="$(arg use_moveit)"/>
      <param name="planning_group_name"  value="$(arg planning_group_name)"/>
      <param name="control_period"       value="$(arg control_period)"/>
      <param name="moveit_sample_duration"  value="$(arg moveit_sample_duration)"/>
  </node>

</launch>

shiba-8ro, 2022-08-10:
@padre221 
Hi,
ID15 (gripper) is not displayed.
So it is possible that the ID15 XM430 settings are incorrect or the servo connection cable is disconnected.

padre221, 2022-08-10:
@shiba-8ro 
Hello shiba
I am aware that the gripper is not detected, because it does not exist, as i stated in the first comment. I only have 4 XM430 setup on the robot without the gripper. What can i do to control it as it is ?

shiba-8ro, 2022-08-14:
@padre221 
I understand now your question.
OpenCR's FW and open_manipulator_controller will not work unless all five XM430 are connected.
So If you wish use only Four XM430, you need to modify both OpenCR FW and open_manipulator_controller source.
Sorry, I'm not familiar with source modification.

padre221, 2022-08-15:
@shiba-8ro 
Thank you for your response, and who do you think i should contact about my problem ? I have already tried mailing @ROBOTIS-Will but unfortunately he did not respond.

yun-goon, 2024-12-19:
@padre221 Apologies for the delayed response. We have assembled a TF team to address OpenManipulator-x issues and are actively working on resolutions. Please note that we do not support EOL (End-of-Life) ROS versions; our support currently extends to Noetic and Humble versions. We kindly ask that you try again with the latest version. If the same issue arises, please feel free to resubmit it, and we will address it promptly.

closed 2024-12-19 by yun-goon